ethz-asl/kalibr #442: _tkinter.TclError: couldn't connect to display

harezhh, 2021-06-22:
Hello,

I am currently working with the Kalibr tool and using the sample dataset to get to know the tool. I am using the Docker installation. I now have a problem at the end of the calibration process. The calibration actually works because the results are written to the appropriate files. However, the GUI with the statisics is not displayed to me and I get the following error message, which can be seen on the image. Unfortunately, I have not been able to solve the problem on my own.

[image]

goldbattle, 2022-04-22:
You need to allow for docker to pass the x11 session to the host to display the pdf report.
For [example](https://github.com/goldbattle/kalibr#docker):
```
xhost +local:root
docker run -it -e "DISPLAY" -e "QT_X11_NO_MITSHM=1" \
    -v "/tmp/.X11-unix:/tmp/.X11-unix:rw" \
    -v "$FOLDER:/data" \
    kalibr
```

Otherwise you can disable the report with `--dont-show-report`.

closed 2022-04-22 by goldbattle

erichliang, 2024-03-31:
I have tried turning on the flag `--dont-showreport` but still get the same error. My docker is unable to generate and save the PDF report, even when trying to skip showing it on screen. Any suggestions on what to do?