ethz-asl/kalibr #445: IMU-Camera extrinsic initial guess

jdgreeno, 2021-06-29:
Hi All,

When using `kalibr_calibrate_imu_camera`, I would like to input an initial guess for the `T_cam_imu` extrinsics for my cameras. The [comment here](https://github.com/ethz-asl/kalibr/blob/e9453e0734abec221d4335b1d6fb3202b1c9ff2b/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py#L107) implies that it is possible, but below the code seems to run `IccCamera.findOrientationPriorCameraToImu()` without any check. It appears as though the hooks to exctract `T_cam_imu` from a `camchain.yaml` are present, but not used during this calibration.

I hope I am not missing something obvious. Has anyone had success doing this? Does it require modifying the code to take the input and skip `IccCamera.findOrientationPriorCameraToImu()`?

Thank you

RuanJiaJun, 2021-12-16:
Since the calibration translation is far from the real value, I also want to input an initial value when the camera and IMU are jointly calibrated like you. Have you realized it now? Can you share the solution? Thank you!

wzjscut, 2022-05-18:
have you solved this problem?

goldbattle, 2022-05-18:
This isn't possible right now (see the tag for this issue), but you could
edit the code yourself to try to do this.