ethz-asl/kalibr #448: kalibr_bagcreater ImportError

bereze, 2021-07-01:
I used kalibr_bagcreater as suggested in wiki, however, it comes the ImportError of ImageFile:
`importing libraries
Traceback (most recent call last):
  File "/home/luo/kalibr_workspace/devel/bin/kalibr_bagcreater", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/luo/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_bagcreater", line 8, in <module>
    import ImageFile
ImportError: No module named ImageFile`
How to solve this problem?Please help.

floriantschopp, 2021-07-01:
Are you on the master branch?
You can try changing it to
```python
from PIL import ImageFile
```

bereze, 2021-07-01:
It works, thank you very much. 
Is it related to the branch that I downloaded?

floriantschopp, 2021-07-01:
Yeah, I think on the master branch it is already fixed.

closed 2021-07-01 by floriantschopp