ethz-asl/kalibr #449: ImportError: No module named rosbag

HerdGold3, 2021-07-05:
When I want to calibrate the d435i, the error is as follows：
anderath@anderath-OMEN-by-HP-Laptop:~$ kalibr_calibrate_cameras --target ~/kalibr_ws/src/Kalibr/data/checkerboard.yaml --bag ~/kalibr_ws/src/Kalibr/data/camd435i.bag --models pinhole-equi --topics /color --show-extractio
importing libraries
Traceback (most recent call last):
  File "/home/anderath/kalibr_ws/devel/bin/kalibr_calibrate_cameras", line 6, in <module>
    exec(fh.read())
  File "<string>", line 3, in <module>
ImportError: No module named rosbag
Can anyone help me?
thanks!

hankhsueh, 2021-07-08:
Maybe you can try this "conda install -c conda-forge ros-rosbag". 
Refer  from : [https://stackoverflow.com/questions/57254026/modulenotfounderror-no-module-named-rosbag-with-pip](url)

HerdGold3, 2021-07-09:
Hi,hankhsudh
First, thank you very much for answering my question
I install rosbag according to your method, but the terminal has been reporting an error cycle, as shown in the figure.
[image]
And the link you gave me is blank after I open it.
I don't know what to do now.

goldbattle, 2022-04-22:
You need to make sure you source your ros environment. I recommend running through `rosrun kalibr <your_cmd>`. Please follow the official install instructions (and PR https://github.com/ethz-asl/kalibr/pull/515)

closed 2022-05-03 by goldbattle