ethz-asl/kalibr #451: Some questions about  calibration result

hankhsueh, 2021-07-06:
Hi all,

I have some questions about calibration results.
Please give me a hand.

My questions are as follows.
# Q1: How to evaluate the calibration result is good or bad?
I know that the smaller residuals are better. Are there any others?
result : [report-imucam-cam.pdf](https://github.com/ethz-asl/kalibr/files/6768624/report-imucam-cam.pdf)

# Q2: What is the relationship between Kalibr's camera coordinate system and IMU's coordinate system? 
Is it like the picture? I really confused between both of them.
[image]

# Q3: My translation vector(T_ci, T_ic) of the result is not matched with the real world. How can I improve it?
The relationship between IMU and camera, like
[image]

hankhsueh, 2021-11-10:
Thanks to both @mhyoosefian and @floriantschopp reply.
 
In order to improve the result by your suggestions, I did some action:
1. Got the own IMU config by Allan analysis(record 2 hrs)
2. Increase camera FPS to 20FPS
3. Make sure the IMU data is correct and sample rate 200Hz( each data interval ~5ms )
4. Use imu-models (scale-misalignment & scale-misalignment-size-effect) to do calibrate. command like :  kalibr_calibrate_imu_camera --target 6x6.yaml --bag cam.bag --imu imu.yaml --perform-synchrozation --imu-models scale-misalignment
5. Install camera and IMU on the robot arm (Avoid shaking) and excite IMU all axes

I got these results.
##Calibrated IMU by myself.
[report-imucam-cam_Calibrated.pdf](https://github.com/ethz-asl/kalibr/files/7511732/report-imucam-cam_Calibrated.pdf)

##Calibrate IMU by using Kalibr imu-models
scale-misalignment:
[report-imucam-cam_scale-misalignment.pdf](https://github.com/ethz-asl/kalibr/files/7511741/report-imucam-cam_scale-misalignment.pdf)

scale-misalignment-size-effect:
[report-imucam-cam_scale-misalignment-size-effect.pdf](https://github.com/ethz-asl/kalibr/files/7511747/report-imucam-cam_scale-misalignment-size-effect.pdf)

I think I got the correct gravity vector close to -9.8, but the translation between the IMU and the camera is too close.

##So I still have some questions:
Q1 : How can I improve the translation T_ci (still too small) ?
Q2 : What is the accuracy of the kalibr_calibrate_imu_camera(T_ci) (error < 1cm?, < 10cm? or others) ?
Q3 : Why do the comparison of predicted and measured specific force/angular velocities sometimes not match the raw data?

Best regards.

mhyoosefian, 2021-11-16:
Hi there,

As @floriantschopp mentioned, you still need to work on your data acquisition procedure. You may want to do smoother/slower movements, while making sure to excite all IMU axes. I have attached the result of a successful calibration. Have a look at its movements (especially error plots). Furthermore, I recommend to download the EuRoC calibration dataset and notice the movements using its images.

Regards
[report-imucam-data.pdf](https://github.com/ethz-asl/kalibr/files/7548882/report-imucam-data.pdf)

zoldaten, 2022-09-29:
@mhyoosefian could you have a look at my calibration pdf please - 
[320x240-report-imucam.pdf](https://github.com/ethz-asl/kalibr/files/9678351/320x240-report-imucam.pdf)
i calibrated raspberry pi fish eye camera with mpu-650.

goldbattle, 2024-08-12:
Feel free to reopen (or create new issue) if there are any lingering concerns.

closed 2024-08-12 by goldbattle