ethz-asl/kalibr #455: Number of threads used for multithreading

Buanderie, 2021-08-05:
Hi,
I was wondering if it was possible to limit the number of simultaneous threads used, as my machine has a 16 cores CPU, and I can see that kalibr is using all of them. Problem is, my memory fills up pretty quickly (16GB available)...
Using a lower number of thread would help me solve this issue... Another strategy would be to add more RAM, also :/

goldbattle, 2022-04-22:
You can see this issue https://github.com/ethz-asl/kalibr/issues/342 on where you can change it. Currently this is a hardcoded ceres parameter for the CAM-IMU optimization problem.

goldbattle, 2022-05-03:
This is actually due to the backend solver `BlockCholeskyLinearSystemSolver` not supporting multiple threads. It would be nice to do what is described in it:
https://github.com/ethz-asl/kalibr/blob/e9453e0734abec221d4335b1d6fb3202b1c9ff2b/aslam_optimizer/aslam_backend/src/BlockCholeskyLinearSystemSolver.cpp#L58-L72

goldbattle, 2022-05-04:
Track via issue #342

closed 2022-05-04 by goldbattle