ethz-asl/kalibr #459: AttributeError: 'AslamCamera object has no attribute 'geometry'

Enes1097, 2021-08-24:
Thanks for providing this package, it seems to be very useful-

I was just trying out the Camera-IMU Calibration with a **Rolling-Shutter Camera and an IMU**, but encountered some problems.
I read through a few issues already talking about calibrating a rolling-shutter camera with an imu instead of a global shutter camera and realized that it's harder to do than it seems. I also tried to implement every hint mentioned in these issues (f.e. minimum exposure time, high fps, slow movements), but it still didn't work.

This is what my system puts out after the starting the calibration:
[image]

I suspect that the Corner Detection of the Aprilgrid doesn't work with the images my camera puts out, that's why I attached an example picture from the rosbag here: 
[image]

P.S.: The Camera I'm using is an Azure Kinect DK Camera which makes use of a **rational-polynomial distortion model**. This model is not supported by kalibr which is why I was using the famous _image-proc_ package to undistorted it and then fill the rosbag with the monochrome undistorted image (/rgb/image_mono) which the _image_proc_ puts out.

If anyone could help me, I would really appreciate it.

Enes1097, 2021-12-13:
> Hi,
> 
> Have you tried calibrating your camera-IMU system, using the **global shutter** process? Although it is not perfect, it must give you acceptable results.
> 
> I faced the error you encountered quite a while ago, but the problem was that some of the images were not recorded successfully, which does not seem to be your case.
> 
> Regards

Hey,

The only camera that I was able to use in that period of time was the Azure Kinect DK camera, which has a rolling-shutter sensor. That way I couldn't make use of the global shutter process, as you pointed out. 

By using this package [https://github.com/JzHuai0108/kalibr](url), a Kalibr ROS-Package for RS-cameras, I did not get this error message and the calculation succeeded. 

Cheers

mhyoosefian, 2021-12-13:
> > Hi,
> > Have you tried calibrating your camera-IMU system, using the **global shutter** process? Although it is not perfect, it must give you acceptable results.
> > I faced the error you encountered quite a while ago, but the problem was that some of the images were not recorded successfully, which does not seem to be your case.
> > Regards
> 
> Hey,
> 
> The only camera that I was able to use in that period of time was the Azure Kinect DK camera, which has a rolling-shutter sensor. That way I couldn't make use of the global shutter process, as you pointed out.
> 
> By using this package https://github.com/JzHuai0108/kalibr, a Kalibr ROS-Package for RS-cameras, I did not get this error message and the calculation succeeded.
> 
> Cheers

Glad to hear that you successfully ran the Kalibr. I have done the calibration using the global-shutter scheme, even though my camera was a rolling-shutter. However, it worked properly.