ethz-asl/kalibr #460: Optimization is killed

LiangtaoZhong, 2021-08-30:
Can anyone tell me this problem? When I am optimizing our cam and imu, I get  this problem:

> kalibr_calibrate_imu_camera --target aprilgrid_6x6.yaml --cam camchain.yaml --imu imu.yaml --bag imu_camera_cali.bag --timeoffset-padding 100
importing libraries
Initializing IMUs:
  Update rate: 167.0
  Accelerometer:
    Noise density: 0.115456108099 
    Noise density (discrete): 1.4920217337 
    Random walk: 0.000858419865715
  Gyroscope:
    Noise density: 0.00624048345999
    Noise density (discrete): 0.0806448190959 
    Random walk: 4.06383605411e-05
Initializing imu rosbag dataset reader:
	Dataset:          imu_camera_cali.bag
	Topic:            /mavros/imu/data_raw
	Number of messages: 12281
Reading IMU data (/mavros/imu/data_raw)
  Read 12281 imu readings over 73.0 seconds                          
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.022 [m]
    Spacing 0.001144 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [1216.99925, 1215.89463]
  Principal point: [326.70365, 229.70743]
  Distortion model: radtan
  Distortion coefficients: [-0.59032, 0.421852, 0.00071, 0.002262]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          imu_camera_cali.bag
	Topic:            /usb_cam/image_raw
	Number of images: 1241
Extracting calibration target corners
  Extracted corners for 101 images (of 1241 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 100
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (sigma): -1.000000
	Gyroscope Huber width (sigma): -1.000000
	Do time calibration: True
	Max iterations: 30
	Time offset padding: 100.000000
Estimating time shift camera to imu:

Initializing a pose spline with 7197 knots (100.000000 knots per second over 71.965931 seconds)
  Time shift camera to imu (t_imu = t_cam + shift):
-1.93080734741

Estimating imu-camera rotation prior

Initializing a pose spline with 7197 knots (100.000000 knots per second over 71.965931 seconds)
Gravity was intialized to [-3.36555931  4.15563129  8.22022883] [m/s^2]
  Orientation prior camera-imu found as: (T_i_c)
[[ 0.85989507  0.26545391  0.43602143]
 [-0.43730426  0.82369943  0.36094909]
 [-0.26333525 -0.50105237  0.8243792 ]]
  Gyro bias prior found as: (b_gyro)
[-0.00708295  0.00732336  0.01723884]

Initializing a pose spline with 47197 knots (100.000000 knots per second over 471.965931 seconds)

Initializing the bias splines with 23598 knots

Adding camera error terms (/usb_cam/image_raw)
  Added 101 camera error terms                                

Adding accelerometer error terms (/mavros/imu/data_raw)
  Added 12281 of 12281 accelerometer error terms (skipped 0 out-of-bounds measurements)

Adding gyroscope error terms (/mavros/imu/data_raw)
  Added 12281 of 12281 gyroscope error terms (skipped 0 out-of-bounds measurements)

Before Optimization
===================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 4.33357380176, median 4.02248979489, std: 3.02166249502
Gyroscope error (imu0):        mean 6.77804179068, median 5.42674952401, std: 6.10045011654
Accelerometer error (imu0):    mean 2.51483642974, median 2.04015624295, std: 2.24747533699

Residuals
----------------------------
Reprojection error (cam0) [px]:     mean 4.33357380176, median 4.02248979489, std: 3.02166249502
Gyroscope error (imu0) [rad/s]:     mean 0.546613954034, median 0.437639233642, std: 0.491969696052
Accelerometer error (imu0) [m/s^2]: mean 3.75219060988, median 3.04395745464, std: 3.35328204875

Optimizing...
Using the block_cholesky linear system solver
Using the levenberg_marquardt trust region policy
Using the block_cholesky linear system solver
Using the levenberg_marquardt trust region policy
Initializing
Optimization problem initialized with 94412 design variables and 33775 error terms
The Jacobian matrix is 92110 x 424839
[0.0]: J: 1.41804e+06
Killed
root@2dc724633b73:/home# kalibr_calibrate_imu_camera --target aprilgrid_6x6.yaml --cam camchain.yaml --imu imu.yaml --bag imu_camera_cali.bag --timeoffset-padding 100
importing libraries
Initializing IMUs:
  Update rate: 167.0
  Accelerometer:
    Noise density: 0.05 
    Noise density (discrete): 0.646142399166 
    Random walk: 0.001
  Gyroscope:
    Noise density: 0.02
    Noise density (discrete): 0.258456959666 
    Random walk: 4.06383605411e-05
Initializing imu rosbag dataset reader:
	Dataset:          imu_camera_cali.bag
	Topic:            /mavros/imu/data_raw
	Number of messages: 12281
Reading IMU data (/mavros/imu/data_raw)
  Read 12281 imu readings over 73.0 seconds                          
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.022 [m]
    Spacing 0.001144 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [1216.99925, 1215.89463]
  Principal point: [326.70365, 229.70743]
  Distortion model: radtan
  Distortion coefficients: [-0.59032, 0.421852, 0.00071, 0.002262]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          imu_camera_cali.bag
	Topic:            /usb_cam/image_raw
	Number of images: 1241
Extracting calibration target corners
  Extracted corners for 101 images (of 1241 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 100
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (sigma): -1.000000
	Gyroscope Huber width (sigma): -1.000000
	Do time calibration: True
	Max iterations: 30
	Time offset padding: 100.000000
Estimating time shift camera to imu:

Initializing a pose spline with 7197 knots (100.000000 knots per second over 71.965931 seconds)
  Time shift camera to imu (t_imu = t_cam + shift):
-1.93080734741

Estimating imu-camera rotation prior

Initializing a pose spline with 7197 knots (100.000000 knots per second over 71.965931 seconds)
Gravity was intialized to [-3.38021881  3.92692406  8.32597208] [m/s^2]
  Orientation prior camera-imu found as: (T_i_c)
[[ 0.86022179  0.26276891  0.43700226]
 [-0.42827529  0.83745584  0.33948194]
 [-0.27676479 -0.47918703  0.83293519]]
  Gyro bias prior found as: (b_gyro)
[-0.00710004  0.00783423  0.01698298]

Initializing a pose spline with 47197 knots (100.000000 knots per second over 471.965931 seconds)

Initializing the bias splines with 23598 knots

Adding camera error terms (/usb_cam/image_raw)
  Added 101 camera error terms                                

Adding accelerometer error terms (/mavros/imu/data_raw)
  Added 12281 of 12281 accelerometer error terms (skipped 0 out-of-bounds measurements)

Adding gyroscope error terms (/mavros/imu/data_raw)
  Added 12281 of 12281 gyroscope error terms (skipped 0 out-of-bounds measurements)

Before Optimization
===================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 4.34161527114, median 4.02330575981, std: 3.07392302478
Gyroscope error (imu0):        mean 2.11436534108, median 1.69357802931, std: 1.90413996961
Accelerometer error (imu0):    mean 5.79651902452, median 4.70823566474, std: 5.1873380977

Residuals
----------------------------
Reprojection error (cam0) [px]:     mean 4.34161527114, median 4.02330575981, std: 3.07392302478
Gyroscope error (imu0) [rad/s]:     mean 0.546472437679, median 0.437717028412, std: 0.492138227324
Accelerometer error (imu0) [m/s^2]: mean 3.74537670932, median 3.04219068825, std: 3.35175908373

Optimizing...
Using the block_cholesky linear system solver
Using the levenberg_marquardt trust region policy
Using the block_cholesky linear system solver
Using the levenberg_marquardt trust region policy
Initializing
Optimization problem initialized with 94412 design variables and 33775 error terms
The Jacobian matrix is 92110 x 424839
[0.0]: J: 1.10319e+06
Killed

pkrush, 2021-08-31:
Maybe this is the issue: _Extracted corners for 101 images (of 1241 images)_. You should be able to extracted corners in almost every image. Maybe get closer? If you use the --verbose flag you can see the images that the April grid is not being found.

goldbattle, 2022-04-22:
It looks like you don't have good extraction, and your time offset between your two sensors is likely unrealistic. I recommend you inspect your raw data and their timestamps. The problem can be killed if your computer runs out of memory, so I would also inspect this.

closed 2022-05-19 by goldbattle