ethz-asl/kalibr #461: test the dataset"imu-camera-calibration"

Liuwenqin-hub, 2021-09-04:
when i use the dataset "imu-camera-calibration"to test whether i install the kalibr successfully or not, the following error occurred:
Initializing IMUs:
Traceback (most recent call last):
  File "/home/liu/work/kalibr_ws/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 245, in <module>
    main()
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 149, in main
    imuConfig = kc.ImuParameters(imu_yaml)
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 426, in __init__
    ParametersBase.__init__(self, yamlFile, "ImuConfig", createYaml)
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 200, in __init__
    self.data = self.readYaml()
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 211, in readYaml
    self.raiseError( "Could not read configuration from {0}".format(self.yamlFile) )
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 234, in raiseError
    raise RuntimeError( "{0}{1}".format(header, message) )
RuntimeError: [ImuConfig Reader]: Could not read configuration from imu_adis16448.yaml
liu@liu-HP-ENVY-Laptop-13-aq1xxx:~/work/kalibr_ws$ kalibr_calibrate_imu_camera --target /home/liu/work/kalibr_ws/dataTest/dynamic/april_6x6.yaml --cam /home/liu/work/kalibr_ws/dataTest/dynamic/camchain.yaml --imu /home/work/kalibr_ws/dataTest/dynamic/imu_adis16448.yaml --bag /home/liu/work/kalibr_ws/dataTest/dynamic/dynamic.bag --bag-from-to 5 45
importing libraries
Initializing IMUs:
Traceback (most recent call last):
  File "/home/liu/work/kalibr_ws/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 245, in <module>
    main()
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 149, in main
    imuConfig = kc.ImuParameters(imu_yaml)
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 426, in __init__
    ParametersBase.__init__(self, yamlFile, "ImuConfig", createYaml)
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 200, in __init__
    self.data = self.readYaml()
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 211, in readYaml
    self.raiseError( "Could not read configuration from {0}".format(self.yamlFile) )
  File "/home/liu/work/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 234, in raiseError
    raise RuntimeError( "{0}{1}".format(header, message) )
RuntimeError: [ImuConfig Reader]: Could not read configuration from /home/work/kalibr_ws/dataTest/dynamic/imu_adis16448.yaml

then i guess the configuration file path is wrong, i changed it. However, the error still existed. can you give me some suggestions?

goldbattle, 2022-04-22:
It looks like the path to the config file is incorrect. Please make sure it is valid.
Please post the config you used if you still have issues.

For example:
```
rostopic: /imu0
update_rate: 200.0 #Hz
accelerometer_noise_density: 0.01 #continous
accelerometer_random_walk: 0.0002 
gyroscope_noise_density: 0.005 #continous
gyroscope_random_walk: 4.0e-06
```

closed 2022-04-22 by goldbattle

cloudlakecho, 2022-05-07:
@chongjeelee Was you able to get the dataset? If yes, would you share with me? Thanks.

Liuwenqin-hub, 2022-05-07:
> @chongjeelee Was you able to get the dataset? If yes, would you share with me? Thanks.

The datasets can be downloaded from “wiki/downloads” menu on this page.