ethz-asl/kalibr #462: kalibr_camera_validator not wokring

kudo1026, 2021-09-08:
Hi,

I'm using the docker image to run the kalibr. I get the calibration results, but when I want to do kalibr_camera_validator, this error happens. Is there anything I did wrong, or is there anyway I could solve this?

```
root@b27b05b70b2d:~/data/euroc# kalibr_camera_validator --cam kalibr_eucm/camchain-cam_april_384x256.yaml --target april_6x6.yaml 
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.088 [m]
    Spacing 0.0264 [m]
initializing camera geometry
Camera /cam0/image_raw:
  Camera model: eucm
  Focal length: [235.87881456277438, 245.64223825061669]
  Principal point: [186.71061987443053, 132.8633029558988]
  EUCM alpha: 0.594491066437
  EUCM beta: 1.12030931874
  Distortion model: none
  Distortion coefficients: []
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
Traceback (most recent call last):
  File "/kalibr_workspace/devel/bin/kalibr_camera_validator", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_validator", line 480, in <module>
    chain_validator = CameraChainValidator(camchain, targetConfig)
  File "/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_validator", line 76, in __init__
    monovalidator = MonoCameraValidator(camConfig, targetParams)
  File "/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_validator", line 389, in __init__
    self.undistorter = self.camera.undistorterType(self.camera.geometry, cv2.INTER_LINEAR, alpha, scale)
AttributeError: 'AslamCamera' object has no attribute 'undistorterType'

```

goldbattle, 2022-05-04:
Can you use the latest docker images to see if this is still an issue for you? If so can you provide a way to reproduce this? Thanks

goldbattle, 2022-09-14:
Closing due to age, please reopen if still an issue.

closed 2022-09-14 by goldbattle