ethz-asl/kalibr #463: Running calibration command on Ubuntu 16.04 aving installed ROS Kinetic and python 2.7. But get run time errors.

laibazahid26, 2021-09-10:
When I write the calibration command given below: 
`/kalibr_calibrate_cameras --target ~/Downloads/april_6x6_80x80cm.yaml --bag ~/Downloads/CalibrationBag.bag --models pinhole-equi --topics camera/image_raw
`
I get the following errors:

```
importing libraries
Initializing cam0:
	Camera model:	  pinhole-equi
	Dataset:          /root/Downloads/CalibrationBag.bag
	Topic:            camera/image_raw
	Number of images: 2086
Extracting calibration target corners
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_cameras", line 5, in <module>
    exec(fh.read())
  File "<string>", line 444, in <module>
  File "<string>", line 175, in main
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/TargetExtractor.py", line 73, in extractCornersFromDataset
    raise RuntimeError("Exception during multithreaded extraction: {0}".format(e))
RuntimeError: Exception during multithreaded extraction: '_sensor_msgs__CompressedImage' object has no attribute 'encoding'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_cameras", line 5, in <module>
    exec(fh.read())
  File "<string>", line 444, in <module>
  File "<string>", line 175, in main
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/TargetExtractor.py", line 73, in extractCornersFromDataset
    raise RuntimeError("Exception during multithreaded extraction: {0}".format(e))
RuntimeError: Exception during multithreaded extraction: '_sensor_msgs__CompressedImage' object has no attribute 'encoding'

``` 

How can I solve these bugs? Any help would be highly appreciated.

ConnaLu, 2021-09-13:
I get the same error,do you get any idea?

laibazahid26, 2021-09-13:
> I get the same error,do you get any idea?

Sorry, not yet.

mhyoosefian, 2021-09-19:
Can you upload the readings of your IMU? However, I recommend using Ubuntu 14.04 and ROS indigo, on which I myself managed to use the Kalibr successfully.

thejerrycheng, 2021-12-15:
hi, is it possible to share the april_6x6_80x80cm.yaml to me. the original folder needs permission to access now

seajayshore, 2022-03-15:
Did you manage to solve this? I have a similar error...

goldbattle, 2022-04-22:
This is due to using a compressed image topic. Currently the repo does not have support for such topic. You can see PR https://github.com/ethz-asl/kalibr/pull/493

goldbattle, 2022-05-08:
Addressed via #523

closed 2022-05-08 by goldbattle