ethz-asl/kalibr #465: Contradiction between frames in the paper and the readings of the sensors

mhyoosefian, 2021-09-19:
Hi. Thanks for the contribution. 

I am a bit confused about the various frames of the EuRoC sensor setup. In the [EuRoC paper](https://journals.sagepub.com/doi/abs/10.1177/0278364915620033) the frames are described as:
`X_I = -Y_C;
Y_I = X_C;
Z_I = Z_C`
(`I` is for the IMU0 and `C` is for the CAM0) which results in the following rotation matrix:
`R_IC =[0 -1 0; 1 0 0; 0 0 1]`
which is totally compatible with the calibration result provided in the yaml file of the camera.
So, my question is that if the `X` axis of the IMU0 is upward, then the accelerometer reading in this axis (when holding the platform nearly vertical) should be about `-9.8`, but the actual readings (according to dataset) are about `9.8` which means the `X` axis is downward!

Any ideas are appreciated!

NikolausDemmel, 2021-09-24:
Hm... I think if the accelerometer axis is pointing "up" and the IMU is standing still w.r.t. earth, a reading of 1g is expected, not -1g. Gravity is pulling down, but the corresponding acceleration is up. (Consider a rocket launching from earth with 1g acceleration. A person inside the rocket will feel 2g gravity pulling down.)

mhyoosefian, 2021-09-24:
T

> Hm... I think if the accelerometer axis is pointing "up" and the IMU is standing still w.r.t. earth, a reading of 1g is expected, not -1g. Gravity is pulling down, but the corresponding acceleration [...]

Thanks for your reply. Well, consider a ball thrown upward, its acceleration is always `1g` downward, so if a frame attached to the ball has say a `z` axis pointing up, the ball acceleration will be `-1g` in that frame, right? Am I missing something?

Regards

NikolausDemmel, 2021-09-24:
It will be in free fall and the accelerometer will read 0. The way you arrive at `-1g` net acceleration w.r.t. earth is by subtracting the acceleration due to gravity, `+1g`, so you get `0 - 1g = -1g`.

mhyoosefian, 2021-09-24:
> It will be in free fall and the accelerometer will read 0. The way you arrive at `-1g` net acceleration w.r.t. earth is by subtracting the acceleration due to gravity, `+1g`, so you get `0 - 1g = -1g`.

So, I think one can write the equation:
`IMU_Reading = Object Acceleration - Gravity`
in any frame. In the ball example, we can write for the `z` axis of the ball frame (which is upward)
`IMU_Reading = -1g - (-1g) = 0`.
And for the IMU case, if the IMU is at rest, its acceleration is zero, so, for the `x` axis which is upward, we can write:
`IMU_Reading = 0 - (-1g) = 1g`.
I think I confused the `IMU_Reading` with the `Object Acceleration`.

closed 2021-09-25 by mhyoosefian