ethz-asl/kalibr #466: package 'igraph' is deprecated

CogmanCVN, 2021-09-22:
Hello, I encountered the following problems when using the **kalibr_calibrate_cameras** tool:
kalibr_calibrate_cameras --target april_6x6.yaml --bag static.bag --models pinhole-equi pinhole-equi omni-radtan omni-radtan --topics /cam0/image_raw /cam1/image_raw /cam2/image_raw /cam3/image_raw

Traceback (most recent call last):
  File "/home/zqy/kalibr_workspace/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/zqy/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 12, in <module>
    import kalibr_camera_calibration as kcc
  File "/home/zqy/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/__init__.py", line 4, in <module>
    from ObsDb import *
  File "/home/zqy/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/ObsDb.py", line 6, in <module>
    import igraph
  File "/home/zqy/.local/lib/python2.7/site-packages/igraph/__init__.py", line 8, in <module>
    warn_deprecation()
  File "/home/zqy/.local/lib/python2.7/site-packages/igraph/_utils.py", line 36, in warn_deprecation
    raise RuntimeError("This package is deprecated. See the deprecation notice above.")
RuntimeError: This package is deprecated. See the deprecation notice above.

My environment is Ubuntu 8.04 and python 3.7. I need help, thanks.

mhyoosefian, 2021-09-24:
Hi.
Have you tried 
`sudo apt-get upgrade pytohn-igraph`
?

CogmanCVN, 2021-09-24:
Can't solve...

CrisXin000, 2021-09-29:
Maybe you can try 
`sudo pip uninstall igraph`
`sudo apt install python-igraph`
it is work for me.

Tsingkai, 2021-11-19:
The same exception……QAQ
Everything isn't  work for me!
Please the world Die!

JazzyFeng, 2021-12-13:
> Maybe you can try `sudo pip uninstall igraph` `sudo apt install python-igraph` it is work for me.

You saved my life!

lemoncoffe, 2022-01-05:
> 

It is great ！Thanks！

goldbattle, 2022-04-22:
Hi please try PR https://github.com/ethz-asl/kalibr/pull/515 to see if that solves your issue.

closed 2022-05-03 by goldbattle

kanhao100, 2023-04-08:
> Maybe you can try `sudo pip uninstall igraph` `sudo apt install python-igraph` it is work for me.

Thank you very much!