ethz-asl/kalibr #467: Google drive link not working (09/23/2021)

qizhu8, 2021-09-23:
I was unable to access the dataset from the wiki page. Would someone help me with that? Thank you!

mhyoosefian, 2021-09-24:
Yeah, it seems you should enter your email and ask them to give you the access. However, I did so five days ago, but have not received any feedback yet! By the way, if you want to test the Kalibr on some dataset, I would recommend using EuRoC mav dataset calibration dataset which is downloadable [here](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets). Use cam_april for camera calibration and use imu_april for camera_IMU calibration. You can compare the results of your calibration with the yaml files provided in the dataset in which the ground-truth calibration results are provided. Hope it helps!

Regards

qizhu8, 2021-09-24:
Thank you so much mhyoosefian! I'll try your recommended dataset!

closed 2021-09-24 by qizhu8

wywywy01, 2021-11-17:
> Yeah, it seems you should enter your email and ask them to give you the access. However, I did so five days ago, but have not received any feedback yet! By the way, if you want to test the Kalibr on some dataset, I would recommend using EuRoC mav dataset calibration dataset which is downloadable from [here](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets) (The last row of the table in the middle of the page). Use cam_april for camera calibration and use imu_april for camera-IMU calibration. You can compare the results of your calibration with the yaml files provided in the dataset in which the ground-truth calibration results are provided. Hope it helps!
> 
> Regards

hi， thanks for your advice.
when i use the EuRoC mav dataset, how i can run the scripts like this?

kalibr_calibrate_cameras --bag [filename.bag] --topics [TOPIC_0 ... TOPIC_N] --models [MODEL_0 ... MODEL_N] --target [target.yaml]

mhyoosefian, 2021-11-17:
Hi there,

This is quite straightforward:
`kalibr_calibrate-cameras --bag data.bag --topics /cam0/images_raw --models pinhole-radtan --target april_6x6.yaml`
Of course, the names may differ depending on your rosbag name, its topics, etc. See [this link](https://github.com/ethz-asl/kalibr/wiki/calibrating-the-vi-sensor) for more information.

Regards

wywywy01, 2021-11-17:
> Hi there,
>
> This is quite straightforward: `kalibr_calibrate-cameras --bag data.bag --topics /cam0/images_raw --models pinhole-radtan --target april_6x6.yaml` Of course, the names may differ [...]

thanks very much！
It is succeed.
kalibr_calibrate-cameras --bag data.bag --topics /cam0/image_raw --models pinhole-radtan --target april_6x6.yaml