ethz-asl/kalibr #469: How to reduce time shift?

mhyoosefian, 2021-09-25:
Hi, again!

How can I decrease the time shift between IMU and camera? Currently it is about 50 (ms) which seems to be high.

Thanks in advance!

ryanyej, 2022-03-09:
how did you solve this? 
curious!

xyflzyn, 2022-03-15:
@mhyoosefian I also met the same problem, could you please tell me how do you solve it？

mhyoosefian, 2022-03-16:
@xyflzyn @ryanyej 

Hi. The problem was with the data acquisition approach. The best practice is to synchronize the IMU and camera data such that each IMU data is received precisely in the middle of its corresponding image (i.e., the middle of exposure time).

Regards,
Mohammad

xyflzyn, 2022-03-16:
Excuse me,I met some problems，when I try to synchronize the IMU and camera data. Could you please give me some advice, thanks a lot!

xyflzyn, 2022-03-16:
Excuse me,I met some problems，when I try to synchronize the IMU and camera data. Could you please give me some advice, thanks a lot! @mhyoosefian

floriantschopp, 2022-03-16:
Have a look here:
https://github.com/ethz-asl/versavis
or
https://furgalep.github.io/bib/nikolic_icra14.pdf

mhyoosefian, 2022-03-16:
@xyflzyn 
Your answer highly depends on the sensor setup and the processor that you use. The resources @floriantschopp mentioned are quite useful. However, in my case, I used a completely customized setup and did not even use ROS to record data, and therefore, synchronization wasn't that straightforward. Additionally, you can add the time-shift to your state vector, consider it in your equations, and estimate it in real-time. A very useful paper is [this](https://journals.sagepub.com/doi/abs/10.1177/0278364913515286) article.

closed 2022-04-22 by goldbattle