ethz-asl/kalibr #480: IR camera calibration error:No corners could be extracted

lanzipoi, 2021-11-02:
I can calibrate RGB camera with kalibr,but when I calibrate IR camera,an error occurred
[image]

goldbattle, 2022-08-18:
I would check the `--show-extraction`, but generally an issue might be that the image does not have enough gradient information. You would need to do some sort of pre-processing to be able to handle this. If this is still an issue, feel free to reopen with a bag that we can debug off of, thanks!

closed 2022-08-18 by goldbattle

1zgh, 2024-04-18:
I encountered the same problem. Here is my infrared image.
[image]