ethz-asl/kalibr #481: catkin_make Error：No route to host

JonathanLJH, 2021-11-05:
[image]
My firewall is down，But I still have this problem.How can I solve the problem?
Ask for some help

mhyoosefian, 2021-11-06:
Hi.

Go to this directory: `Home/Kalibr_workspace/src/kalibr/suitesparse`. Open `CmakeLists.txt`. Find this expression: `http://faculty.cse.tamu.edu/davis/SuiteSparse`. Replace it with this one: `https://mirror.sobukus.de/files/src/suitesparse/`. 

This error happens because some files are no longer available on the same server that was used when Kalibr was being developed.

Regards,
Mohammad

JonathanLJH, 2021-11-22:
> Hi.
>
> Go to this directory: `Home/Kalibr_workspace/src/kalibr/suitesparse`. Open `CmakeLists.txt`. Find this expression: `http://faculty.cse.tamu.edu/davis/SuiteSparse`. Replace it with this one: [...]

thank you

JonathanLJH, 2021-11-22:
> Hi.
>
> Go to this directory: `Home/Kalibr_workspace/src/kalibr/suitesparse`. Open `CmakeLists.txt`. Find this expression: `http://faculty.cse.tamu.edu/davis/SuiteSparse`. Replace it with this one: [...]

When I do what you said above, I still get the same report of error
[image]

goldbattle, 2022-04-27:
Please try PR https://github.com/ethz-asl/kalibr/pull/515 which uses the system install of suitesparse.

closed 2022-05-03 by goldbattle