ethz-asl/kalibr #482: Check failed: isValidRotationMatrix(matrix)

Pan-long-ming, 2021-11-15:
I met this error,when i run "rosrun rovioli tutorial_thermal_live save_folder" using my own data,this is my calibration file 

label: "tau2 - original_calibration"
id: 412eab8e4058621f7036b5e765dfe812
cameras:
- camera:
    label: /tau_nodelet
    id: 54812562fa109c40fe90b29a59dd7798
    line-delay-nanoseconds: 0
    image_height: 512
    image_width: 640
    type: pinhole
    intrinsics:
      cols: 1
      rows: 4
      data: [440.381678762, 440.34105427, 328.082910449, 257.335140534]
    distortion:
      type: radial-tangential  
      parameters:
        cols: 1
        rows: 4
        data: [-0.125172452973,-0.0158508045529, 0.00552041769855,
    1.00188160221577]
  T_B_C:
    cols: 4
    rows: 4
    data: [0.0207981429818, -0.9997828123456, 0.0013528123456, -0.019282391666587094,
           -0.3239881123456, -0.0080198123456, -0.9460272123465, -0.007020499072517088,
          0.9458326123456, 0.0192373123456, -0.3240845123465, 0.13646235628270134,
           0.0, 0.0, 0.0, 1.0]

The error is "F1115 11:32:11.243661 18011 rotation-quaternion-inl.h:115] Check failed: isValidRotationMatrix(matrix) 000.0207981 00-0.999783 00.00135281
00-0.323988 -0.00801981 00-0.946027
0000.945833 000.0192373 00-0.324085"

Is there any problem in my calibration file or something wrong,I will very appreciate if you could give 
me some advise.

goldbattle, 2022-04-22:
I suggest you open a issue on the rovioli repo for this as compared to here. Looking at the error you posted, it is complaining that the reported rotation is not valid. You likely need to make it valid if it isn't (it should be an orthonormal matrix), or change the parsing.

closed 2022-05-19 by goldbattle

snakehaihai, 2023-05-02:
It is a known problem. The rotation matrix from Kalibr isn`t orthonormal. Can verify by R*R^T. There are 0.01~0.03 at none diagonal region of R*R^T