ethz-asl/kalibr #483: Some errors occured during catkin build -DCMAKE_BUILD_TYPE=Release -j4

Tsingkai, 2021-11-17:
### Environment
   + OS:       Ubuntu 20.04LTS
   + OpenCV:    3.4.1
   + python3:     3.8
   + python2:     2.7

### Errors

#### 1.~~OpenCV?~~ （Solved by Reference ）

```
kalibr_workspace/src/Kalibr/aslam_offline_calibration/ethz_apriltag2/src/example/apriltags_demo.cpp: In member function ‘void Demo::setup()’:
kalibr_workspace/src/Kalibr/aslam_offline_calibration/ethz_apriltag2/src/example/apriltags_demo.cpp:305:15: error: ‘CV_CAP_PROP_FRAME_WIDTH’ was not declared in this scope
  305 |     m_cap.set(CV_CAP_PROP_FRAME_WIDTH, m_width);
      |               ^~~~~~~~~~~~~~~~~~~~~~~
kalibr_workspace/src/Kalibr/aslam_offline_calibration/ethz_apriltag2/src/example/apriltags_demo.cpp:306:15: error: ‘CV_CAP_PROP_FRAME_HEIGHT’ was not declared in this scope
  306 |     m_cap.set(CV_CAP_PROP_FRAME_HEIGHT, m_height);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~
kalibr_workspace/src/Kalibr/aslam_offline_calibration/ethz_apriltag2/src/example/apriltags_demo.cpp: In member function ‘void Demo::loop()’:
kalibr_workspace/src/Kalibr/aslam_offline_calibration/ethz_apriltag2/src/example/apriltags_demo.cpp:383:39: error: ‘CV_BGR2GRAY’ was not declared in this scope
  383 |       cv::cvtColor(image, image_gray, CV_BGR2GRAY);
```
*I solve this problem by this*
---
```
#include <opencv2/core/core.hpp>
#include <opencv2/highgui/highgui.hpp>
#include <opencv2/highgui/highgui_c.h>
```
***
***
#### 2.~~What's this?~~（Solved by Reference ）
```
/usr/include/python3.8/numpy/__multiarray_api.h:1541:35: error: return-statement with a value, in function returning ‘void’ [-fpermissive]
 1541 | #define NUMPY_IMPORT_ARRAY_RETVAL NULL
      |                                   ^~~~
/usr/include/python3.8/numpy/__multiarray_api.h:1546:151: note: in expansion of macro ‘NUMPY_IMPORT_ARRAY_RETVAL’
 1546 | #define import_array() {if (_import_array() < 0) {PyErr_Print(); PyErr_SetString(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }

```
***
***
#### 3. Sorry I can't,QwQ 
```
kalibr_workspace/src/Kalibr/aslam_cv/aslam_imgproc/include/aslam/implementation/PinholeUndistorter.hpp:112:86: error: could not convert ‘cv::Size_<int>(width, height)’ from ‘cv::Size’ {aka ‘cv::Size_<int>’} to ‘CvSize’
  112 |   idealCameraMatrix = aslamcv_helper::getOptimalNewCameraMatrix<distorted_geometry_t>(
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
      |                                                                                      |
      |                                                                                      cv::Size {aka cv::Size_<int>}
  113 |                                                         _distortedGeometry, cv::Size(width, height),
      |                                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  114 |                                                         alpha, cv::Size(newWidth, newHeight));
      |                                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

### Reference
1. [Ubuntu 20.04 + ROS Noetic + OpenCV 4.10编译kalibr相机标定功能包](https://blog.csdn.net/hanyeguxingwo/article/details/121293315?spm=1001.2014.3001.5501)(This can't solve all errors I met)

SasaKuruppuarachchi, 2021-12-01:
Thank you very much for sharing. I saved a lot of time.

Tsingkai, 2021-12-02:
> Thank you very much for sharing. I saved a lot of time.
> 
> I just needed to do `sudo apt-get install libsuitesparse-dev`

I think that you should use the comands to test the result, such as kelibr_bagcreator .I faild ! I found many python syntax errors and some functions are not included in so files.

best wishes to you!

If you succeed, hope your reply……

goldbattle, 2022-04-22:
Hi please try PR https://github.com/ethz-asl/kalibr/pull/515 to see if that solves your issue.