ethz-asl/kalibr #486: Calibrating depth cameras

oleg-alexandrov, 2021-12-05:
I have a setup consisting of two cameras, an image camera, and an image + depth camera. For the second one, in addition to a float pixel value at (i, j) there is also a depth measurement (x, y, z) for that (i, j). 

I used Kalibr and got a solution for the extrinsics and intrinsics. Regretfully, the solution is not correct since Kalibr apparently cannot incorporate the depth information in the calibration, it uses only the pixel values. With the optical center computed for the depth camera, the depth measurement (x, y, z) does not project back into pixel (i, j), there is actually a more than 10 pixel offset.

I understand that if all cameras are modeled perfectly, and if Kalibr converges to the unique exact solution, in which a measured calibration target point projects perfectly into its image in the cameras, then what I say above should not happen.

As it is, there is always some calibration error, and the solution Kalibr does not use the depth information to inform the solver, with the result that the depth and intensity components of the depth camera are disconnected given the solution Kalibr provides.

In short, can Kalibr use the depth information when calibrating a depth camera? Or it only uses the intensity information?

goldbattle, 2022-06-16:
I don't think kalibr can be used to calibrate a depth camera. It relies on extraction of the april markers from an intensity image [0,255] in value. Hope this helps.

oleg-alexandrov, 2022-07-24:
Thank you. Due to this limitation, I wrote my own rig calibrator which can take into account the depth values. That can be important, because sometimes the depth measurements and the intensity image can be somewhat out of sync, so using the depth values improves the result. My own calibrator is documented here: https://stereopipeline.readthedocs.io/en/latest/tools/rig_calibrator.html. Maybe this can be useful of anybody has a depth sensor which needs calibration.