ethz-asl/kalibr #488: Bug reported for the multi-imu calibration

yangyulin, 2021-12-14:
Hi, 

I have found a bug for the multi-imu calibration. 

The results generated for the T_i_b is not in the right form. 

The rotation part is R_i_b while the position part is p_b_i.

yangyulin, 2022-05-17:
@phil0stine, Yea. the b stands for body or based IMU. 
The T_i_b represents the transformation from body frame {b} to auxiliary IMU frame {i}.  
T_i_b should equals to {R_i_b, p_i_b} ( which transforms a vector from {b} to {I}). 

I think you are correct. In their output results, the rotation is R_i_b (rotation from {b} to {i}), while the position is p_b_i (translation from {i} to {b}). 
The current results of the Kalibr are: 
T_i_b = {R_i_b, p_b_i}.

goldbattle, 2022-08-08:
Fixed via https://github.com/ethz-asl/kalibr/pull/525

closed 2022-08-08 by goldbattle