ethz-asl/kalibr #491: Custom Targets are not being generated

hakunaMatataHub, 2021-12-19:
I am using Master source for generating custom april grids for calibration.
Here is my command on ubuntu 20.04 'python2.7 kalibr_create_target_pdf --type apriltag --nx 3 --ny 3 --tsize 0.10 --tspace 0.2' and i installed pyx with pip install pyx ==0.12.1

I get the following error
Traceback (most recent call last):
  File "kalibr_create_target_pdf", line 208, in <module>
    generateAprilBoard(canvas, parsed.n_cols, parsed.n_rows, parsed.tsize, parsed.tagspacing, parsed.tagfamiliy, parsed.skipIds)
  File "kalibr_create_target_pdf", line 124, in generateAprilBoard
    c.text(pos[0]+tagSize*0.3, pos[1], "x")
  File "/usr/local/lib/python2.7/dist-packages/pyx/canvas.py", line 324, in text
    return self.insert(self.texrunner.text(x, y, atext, *args, **kwargs))
  File "/usr/local/lib/python2.7/dist-packages/pyx/text.py", line 1194, in text
    self.execute(expr, self.defaulttexmessagesdefaultrun + self.texmessagesdefaultrun + texmessages)
  File "/usr/local/lib/python2.7/dist-packages/pyx/text.py", line 951, in execute
    self.defaulttexmessagesstart + self.texmessagesstart)
  File "/usr/local/lib/python2.7/dist-packages/pyx/text.py", line 1005, in execute
    self.texinput.write(self.expr)
IOError: [Errno 32] Broken pipe
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/local/lib/python2.7/dist-packages/pyx/text.py", line 748, in _cleantmp
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/pyx/text.py", line 663, in run
    raise RuntimeError("TeX/LaTeX finished unexpectedly")
RuntimeError: TeX/LaTeX finished unexpectedly

    texrunner.texinput.write("\n\\end\n")
IOError: [Errno 32] Broken pipe
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/local/lib/python2.7/dist-packages/pyx/text.py", line 748, in _cleantmp
    texrunner.texinput.write("\n\\end\n")
IOError: [Errno 32] Broken pipe

I did try to fix this with adding these lines at the kalibr_create_taget_pdf python script:
from signal import signal, SIGPIPE, SIG_DFL
signal(SIGPIPE,SIG_DFL) 
but this would generate a .tex file which is empty 

Any help is appreciated.

goldbattle, 2022-04-22:
Hi please try PR https://github.com/ethz-asl/kalibr/pull/515 to see if that solves your issue.

goldbattle, 2023-02-13:
I was able to run on the master with:
```
rosrun kalibr kalibr_create_target_pdf --type apriltag --nx 3 --ny 3 --tsize 0.10 --tspace 0.2
```

Feel free to reopen if still an issue for you.

closed 2023-02-13 by goldbattle