ethz-asl/kalibr #492: The center y error of the calibration of a single RGB camera is too large

CNSTJiatongLee, 2021-12-23:
When I use 
`kalibr_calibrate_cameras --topics /camera/rgb/image_raw --bag data.bag --models pinhole-radtan`  
to calibrate the monocular rgb camera, I get a wrong center y.
The image resolution is` [1280, 720]` . The intrinsics is `[818.04722, 818.07404, 640.211278, 462.83350]` and the reproduction error is `[0.391431, 0.288629]`. Even though I set the initial projection to `[810, 810, 640, 360]`, the final result is still wrong.

mhyoosefian, 2021-12-26:
Hi,

What do you mean by 'wrong'? i.e., how can you conclude that the result is wrong? Given the relatively high resolution of your images, the reprojection error seems reasonable. However, I recommend using MATLAB/ OpenCV for calibration and comparing the results.

Regards,
Mohammad

ishipachev, 2022-01-04:
It looks like it calibrated to correct values for 1280 х 920 images. What is the resolution of your images?
Are you sure that it wasn't cropped?

goldbattle, 2022-08-18:
Feel free to reopen the issue if needed.

closed 2022-08-18 by goldbattle