ethz-asl/kalibr #494: I am unable to find the April grid pdf File

Avi241, 2022-01-06:
Hello Everyone I am unable to find the April grid pdf file. It says access denied.
I have also tried on this website but still, I am unable to get it. https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets#downloads please if anyone already have it can share it to me.
Thanks in Advance

NikolausDemmel, 2022-01-06:
Those are the files from the wiki that I can download:

* [april_6x6_80x80cm_A0.pdf](https://github.com/ethz-asl/kalibr/files/7821505/april_6x6_80x80cm_A0.pdf)
* [april_6x6_80x80cm.pdf](https://github.com/ethz-asl/kalibr/files/7821506/april_6x6_80x80cm.pdf)

The links used to work for everyone, but then google changed their policy and it seems no one is watching that account to give approvals.

You should be able to generate any pdfs yourself using the script that comes with Kalibr. Might even work in the docker container, that should be very to set up.

Avi241, 2022-01-06:
Thank You so much for the files.
Yaah I have tried to generate the files by the script but I ran into an error that I was unable to solve.
Anyways thanks for the files.

closed 2022-01-06 by Avi241