ethz-asl/kalibr #500: How to solve calibrate_camera, calibrate_imu_camera import error?

YouMinJung, 2022-02-16:
**My environment : Ubuntu 14.04, ROS indigo, Python 2.7**

When I execute the kalibr_calibrate_camera, I get the "ImportError : No module named sm" error message.

To install the "sm", I commanded the "sudo pip install sm" but I got this error message...
`Traceback (most recent call last) : 
File "/usr/local/bin/pip", line 7, in <module>
from pip._internal.cli.main import main
ImportError : No module named pip._internal.cli.main`

How to solve this problem..??

alec-bell, 2022-03-08:
You're probably trying to run the Python scripts directly, but you need to execute them with ROS. Assuming you have already built from source, please run the following commands:
```
cd ~/ws/kalibr
source devel/setup.bash
rosrun kalibr kalibr_calibrate_cameras \
    --bag </path/to/intrinsics/bag> \
    --topics </axis/image_raw> \
    --models <model name> \
    --target </path/to/target.yaml>
```

goldbattle, 2022-06-02:
We don't support 14.04 any more. Please use a docker container if you are on that.

closed 2022-06-02 by goldbattle