ethz-asl/kalibr #503: RuntimeError: Exception during multithreaded extraction: bad numeric conversion: positive overflow

zwjgit, 2022-02-28:
Hello，When I calibrated the camera, the following error occurred：

Extracting calibration target corners
Traceback (most recent call last):
  File "/home/shrc/workspace/01_ros/kalibr_workspace/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/shrc/workspace/01_ros/kalibr_workspace/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 447, in <module>
    main()
  File "/home/shrc/workspace/01_ros/kalibr_workspace/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 178, in main
    noTransformation=True)
  File "/home/shrc/workspace/01_ros/kalibr_workspace/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_common/TargetExtractor.py", line 73, in extractCornersFromDataset
    raise RuntimeError("Exception during multithreaded extraction: {0}".format(e))
RuntimeError: Exception during multithreaded extraction: bad numeric conversion: positive overflow

How do I fix it？Thanks.

trioKun, 2022-05-04:
I met the same problem when running kalibr_calibrate_imu_camera on a 4-core 8GB-RAM laptop, and then succeeded on another 12-core 32GB-RAM machine. So I guess it might relate to a large number of parallel threads or the use of excessive memory.

goldbattle, 2022-06-02:
I think this is due to memory of the platform possibly. Very hard to debug. If anybody is able to reproduce / re-create it then please reopen.

closed 2022-06-02 by goldbattle

vswdigitall, 2022-07-23:
I got same error on Nvidia Jetson NX. 
Not enough RAM.
Solution:
edit kalibr_calibrate_cameras
multithreading = False #not (parsed.verbose or parsed.showextraction)

russelldj, 2022-09-07:
Alternatively you can use the `--verbose` or `--show-extraction` flags to accomplish the same goal without worrying about modifying the source.