ethz-asl/kalibr #505: why my camera seems so laggy in kalibr with--show-extraction ,however ,directly showed in rviz is very smooth

ryanyej, 2022-03-08:
my camera is telephoto lens.
When using --show-extraction ,compare with demo bag, my camera was too laggy, tend to lose track of corners.
And after many tests. Reprojection error report varies from 3pixs to 10pixs, which is really large that i can't accept.
below are one of many debug info:
kalibr_calibrate_imu_camera --target april_6x6.yaml --cam camchain.yaml --imu imu_cuav.yaml --bag 2022-03-07-18-03-58.bag --timeoffset-padding 0.2
importing libraries
Initializing IMUs:
  Update rate: 200.0
  Accelerometer:
    Noise density: 0.0418625607406 
    Noise density (discrete): 0.59202601155 
    Random walk: 0.00397251262469
  Gyroscope:
    Noise density: 0.00324051779349
    Noise density (discrete): 0.0458278421266 
    Random walk: 2.78858134022e-05
Initializing imu rosbag dataset reader:
	Dataset:          2022-03-07-18-03-58.bag
	Topic:            /mavros/imu/data_raw
	Number of messages: 24430
Reading IMU data (/mavros/imu/data_raw)
  Read 24430 imu readings over 123.2 seconds                          
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.08 [m]
    Spacing 0.024 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [4267.879208, 4267.711531]
  Principal point: [1104.506875, 475.962553]
  Distortion model: radtan
  Distortion coefficients: [-0.022947, 0.184962, -0.000188, 0.011088]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          2022-03-07-18-03-58.bag
	Topic:            /usb_cam0/image_raw
	Number of images: 3410
Extracting calibration target corners
  Extracted corners for 3397 images (of 3410 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 100
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (sigma): -1.000000
	Gyroscope Huber width (sigma): -1.000000
	Do time calibration: True
	Max iterations: 30
	Time offset padding: 0.200000
Estimating time shift camera to imu:

Initializing a pose spline with 12318 knots (100.000000 knots per second over 123.177664 seconds)
  Time shift camera to imu (t_imu = t_cam + shift):
-0.17651612786256293

Estimating imu-camera rotation prior

Initializing a pose spline with 12318 knots (100.000000 knots per second over 123.177664 seconds)
Gravity was intialized to [-8.41282901  4.55345235 -2.15842594] [m/s^2]
  Orientation prior camera-imu found as: (T_i_c)
[[ 0.43196849 -0.75739043  0.48965596]
[... 113 lines cut ...]
    Noise density (discrete): 0.59202601155 
    Random walk: 0.00397251262469
  Gyroscope:
    Noise density: 0.00324051779349
    Noise density (discrete): 0.0458278421266 
    Random walk: 2.78858134022e-05
  T_i_b
    [[1. 0. 0. 0.]
     [0. 1. 0. 0.]
     [0. 0. 1. 0.]
     [0. 0. 0. 1.]]
  time offset with respect to IMU0: 0.0 [s]

  Saving camera chain calibration to file: camchain-imucam-2022-03-07-18-03-58.yaml

  Saving imu calibration to file: imu-2022-03-07-18-03-58.yaml
  Detailed results written to file: results-imucam-2022-03-07-18-03-58.txt
Generating result report...
/home/jueleiye/.local/lib/python2.7/site-packages/matplotlib/cbook/deprecation.py:107: MatplotlibDeprecationWarning: Passing one of 'on', 'true', 'off', 'false' as a boolean is deprecated; use an actual boolean (True/False) instead.
  warnings.warn(message, mplDeprecation, stacklevel=1)
/home/jueleiye/.local/lib/python2.7/site-packages/matplotlib/cbook/deprecation.py:107: MatplotlibDeprecationWarning: Adding an axes using the same arguments as a previous axes currently reuses the earlier instance.  In a future version, a new instance will always be created and returned.  Meanwhile, this warning can be suppressed, and the future behavior ensured, by passing a unique label to each axes instance.
  warnings.warn(message, mplDeprecation, stacklevel=1)
/home/jueleiye/kalibr_workspace/src/kalibr/Schweizer-Messer/sm_python/python/sm/PlotCollection.py:57: wxPyDeprecationWarning: Using deprecated class PySimpleApp. 
  app = wx.PySimpleApp()
  Report written to report-imucam-2022-03-07-18-03-58.pdf

goldbattle, 2022-05-04:
Can you provide a bag that has this issue?

ryanyej, 2022-05-04:
> 

Of course.However the bag size is 4.4Gb.
So i don't know how to upload this bag for analysis. @goldbattle

ryanyej, 2022-05-04:
Hi, @goldbattle , I make a compress and the bag is 1.4Gb.
upload in below link, don't know if you can access or not?

link：https://pan.baidu.com/s/1wWK-FQFyWdSLMiPA0nkhXw?pwd=0vy7 
let me know if you have any question, thx!