ethz-asl/kalibr #506: CDE package not longer public

alwynmathew, 2022-03-10:
The [GDrive link](https://drive.google.com/file/d/0B0T1sizOvRsUVDBVclJCQVJzTFk/edit?usp=sharing) to kalibr CDE package is no longer downloadable. The link now redirects to a "Request Access" gdrive page.

JessicaBertolasi, 2022-03-15:
Hi, I have the same problem, how did you fixed it? I cant't download anything!

redchillipadi, 2022-03-21:
I am trying to use the docker image as described in https://github.com/ethz-asl/kalibr/wiki/installation as a work around for now.

JessicaBertolasi, 2022-03-21:
ok thaks! but how do you fix the CDE packages problem? because without them i can't do anything ....

carlosfloar, 2022-03-29:
Hi, I have the same problem to download the CDE, the link redirects to google drive where they report that "Sorry, the file you requested does not exist.", Does anyone know of another source?

alwynmathew, 2022-03-31:
You can use this docker image https://hub.docker.com/r/stereolabs/kalibr

goldbattle, 2022-04-22:
See this issue https://github.com/ethz-asl/kalibr/issues/514

closed 2022-04-22 by goldbattle