ethz-asl/kalibr #507: Apply the distortion model with Kalibr

echo-gui, 2022-03-21:
I try to calibrate the Oxford Robotcar Dataset using kalibr or basalt. Basalt seems to work with aprilgrid when doing the calibration, while the Oxford Robotcar Dataset only provides the calibration sequence with a checkboard. 

So I calibrate the mono-left wide-angle camera with omni-radtan camera model on kalibr. Here is the result I got:

[image]

Then I try to undistort the fisheye images with the distortion coeffs, but the undistorted images seem to be **the same as the original fisheye image:**
[image]

My idea is that undistorted images might be different. Do you have any advice on this? @NikolausDemmel

echo-gui, 2022-03-21:
Here is how I undistort the fisheye images, (x,y) is the pixel position on image coordinates. k1, k2, r1, r2 are the distortion parameters provided by kalibr omni-radtan.

[image]

NikolausDemmel, 2022-04-01:
The omni-radtan model in Kalibr works like this (see also https://github.com/ethz-asl/kalibr/blob/445b9463abbfdc90498193c5fa20b20be70c036d/aslam_cv/aslam_cameras/include/aslam/cameras/implementation/OmniProjection.hpp#L74-L114):
1. Project 3D point onto unit-sphere
2. Generic pinhole projection onto z=1 plane with offset xi
3. Apply radtan distortion
4. Convert to pixels by scaling with focal length and applying center offset

To undistort an image you typically do the following.
1. Define a target model, e.g. a perfect pinhole model w/o distortion (to get images where straight lines are straight lines). In principle, it can be any model you like.
2. Iterate over the pixels of your target image
3. For every pixel position, unproject the target pixel (e.g. multiply with K^-1 for pinhole).
4. Project this unprojected point back onto the original image with the original camera model.
5. Determine the pixel value by interpolation (e.g. bilinear), or some default value like "black" if it's out-of-bounds.

I didn't check your equations exactly, but it seems that at the very least you are missing the spherical projection + offset by xi between line 1 and 2. Note that the K^-1 in your line 1 and the K in the last line don't have do be the same. You can pick a different K^-1 for the target model.

I hope this helps.

echo-gui, 2022-04-02:
Thanks for your reply. It helps a lot! 

I calibrate the camera with Omni+radtan, and I want to get a perfect Omni camera model without the distortion (the rad-tan distortion parameters).
Since the original camera model and the target camera model are the same, I just want to get rid of the rad-tan distortion on the calibration result. Hence, I did not include the spherical projection + offset on the equation. 

I check the code that you link. The code uses the following equations to distort the point on the plane (z=1). 
[image]

If I want to undistort the point to get rid of the rad-tan distortion model, should I apply an inverse of the rad-tan distortion equation (Equ.6 and Equ.7)? which means we got the $x_{distort}$ and $y_{distort}$ and I want the "undistort" $m_x$ and $m_y$. 

I really appreciate your opinion :)

NikolausDemmel, 2022-04-02:
> I calibrate the camera with Omni+radtan, and I want to get a perfect Omni camera model without the distortion (the rad-tan distortion parameters).
Since the original camera model and the target camera model are the same, I just want to get rid of the rad-tan distortion on the calibration result. Hence, I did not include the spherical projection + offset on the equation.

Ah, sorry, I misunderstood. You are right. In this case you can skip the spherical projection part b/c it's the same for original and target model.

> If I want to undistort the point to get rid of the rad-tan distortion model, should I apply an inverse of the rad-tan distortion equation (Equ.6 and Equ.7)? which means we got the $x_{distort}$ and $y_{distort}$ and I want the "undistort" $m_x$ and $m_y$.

Depends on what you mean by "undistort the point". If you want to compute an undistorted image (which you can then use with the unified model (= omni-radtan with 0 distortion parameters)) and you follow the procedure I laid out above ("To undistort an image you typically do the following"), then you only need the forward projection model for the original calibration. You need the inverse function only if your target model also has distortion. You also need the inverse of the distortion function if you want to unproject ("backproject") a point in the original image from pixels to 3d ray. 

> Then I try to undistort the fisheye images with the distortion coeffs, but the undistorted images seem to be the same as the original fisheye image:

Are the images really identical? You can check for example the mean pixel shift you get when computing the undistorted image. You can also see that the distortion parameters are quite small, so the distortion function is close to the identity function. The idea with omni-radtan is that most of the fisheye effect is captured by the spherical projection and only small imperfections are corrected by the additional radtan distortion. So it's not surprising that the two images are similar, but they should not be identical.

echo-gui, 2022-04-02:
> 3. For every pixel position, unproject the target pixel (e.g. multiply with K^-1 for pinhole).
> 4. Project this unprojected point back onto the original image with the original camera model.

Yeah, I did the undistort algorithm exactly as "To undistort an image you typically do the following", except the spherical projection + offset. 

> Are the images really identical? 
>  You can also see that the distortion parameters are quite small, so the distortion function is close to the identity function. The idea with omni-radtan is that most of the fisheye effect is captured by the spherical projection and only small imperfections are corrected by the additional radtan distortion.

They images are not identical, pay attention to the right side boundary. There is an obvious offset of the pixels.
I agree with your opinion. If I set the rad-tan distortion parameters to 0, only using the Omni camera model works fine with the SLAM related algorithms. 

> You need the inverse function only if your target model also has distortion. You also need the inverse of the distortion function if you want to unproject ("backproject") a point in the original image from pixels to 3d ray.

I'm confused here. The kalibr uses the rad-tan equation the distort the point. Why does getting a unified model (= omni-radtan with 0 distortion parameters) not need to inverse the distortion process? 

To get a perfect unified camera model (with 0 distortion parameters), I think first unprojecting the point on the image plane to its viewing ray, then forward projecting the 3d ray to the image plane with the same Omni camera model with 0 distortion parameters. So the inverse distortion process is needed by unprojecting the point on the image plane to 3d ray.

NikolausDemmel, 2022-04-02:
> To get a perfect unified camera model (with 0 distortion parameters), I think first unprojecting the point on the image plane to its viewing ray, then forward projecting the 3d ray to the image [...]

It depends what you want to do. If you want for a given point in the original image (w/ distortion) find the corresponding point in the undistorted image, then yes, you need to invert the distortion function as you describe. But that's not how we compute the undistorted image as a preprocessing step. For that we do the opposite: For each pixel in the undistorted image, we ask which pixel location it corresponds to in the original image (such that we can look up the corresponding color by interpolation). So we only need the distortion function, not the inverse.

echo-gui, 2022-04-02:
> But that's not how we compute the undistorted image as a preprocessing step. For that we do the opposite: For each pixel in the undistorted image, we ask which pixel location it corresponds to in [...]

Now I totally understand what you said, and learned a lot from this conversation.

Thank you for your time! ;)

closed 2022-04-02 by NikolausDemmel