ethz-asl/kalibr #520: Update Wiki Content

goldbattle, 2022-04-27:
- [ ] Download links https://github.com/ethz-asl/kalibr/wiki/downloads
- [ ] Install guide to user Docker https://github.com/ethz-asl/kalibr/wiki/installation
- [ ] Distortion models references https://github.com/ethz-asl/kalibr/wiki/Supported-models
- [ ] Packages to do allan variance https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model
- [ ] Example wiki on calibrating T265 / realsense sensors
- [ ] Laser range finder guide https://github.com/ethz-asl/kalibr/wiki/Camera-IMU-LRF-calibration

alvaro-oliver, 2022-07-06:
Would you explain in "Multi IMU and IMU intrinsic calibration" if is it possible to get intrinsic calibration values for a single IMU with no video reference?. What would be the experimental procedure for doing so?

A feature like that would expand the uses of Kalibr beyond inertial-video researchers, since Alberto Pretto's [IMU_Tk](https://bitbucket.org/alberto_pretto/imu_tk/src/master/) development is long time abandoned.

goldbattle, 2022-07-08:
> Would you explain in "Multi IMU and IMU intrinsic calibration" if is it possible to get intrinsic calibration values for a single IMU with no video reference?. What would be the experimental [...]

I have updated the wiki's second paragraph to state that you need at least one camera.

zoldaten, 2022-09-29:
a make a short post about Kalibr if you dont mind - https://habr.com/ru/post/689778/

sgraybi, 2023-04-04:
@goldbattle maybe the attached document will help with "How to use IMU intrinsic calibration results".  Please verify for correctness!
[githubreply.md](https://github.com/ethz-asl/kalibr/files/11151500/githubreply.md)

goldbattle, 2023-04-04:
> @goldbattle maybe the attached document will help with "How to use IMU intrinsic calibration results". Please verify for correctness! [KalibrHowToApplyCalibration.md](https://github.com/ethz-asl/kalibr/files/11151798/KalibrHowToApplyCalibration.md)

Great write up. Are you ok with me incorporating this into a wiki page for others to reference?

sgraybi, 2023-04-04:
> Great write up. Are you ok with me incorporating this into a wiki page for others to reference?

Yes, please do.  That was the intention.  Please verify the document for accuracy.  I believe I got the details correct but it would be great to get an expert review.

sgraybi, 2023-07-07:
@goldbattle do you have any update on "How to use IMU intrinsic calibration results".  I have some accuracy issues in my system.  If I can rule out that I'm applying the calibration correctly that would help a lot!  Please feel free to link to other libraries or documentation if that is easier.