ethz-asl/kalibr #524: What Is "external_trigger" Input Variable in Laser Calibration (Related List of Header in CSV File)?

cloudlakecho, 2022-05-10:
Thanks for putting detailed instruction for [Laser Calibration](https://github.com/ethz-asl/kalibr/wiki/Camera-IMU-LRF-calibration). 

I have question on "external_trigger" input variable in the code line:  
`reader = kc.BagTriggeredScanDatasetReader(bagfile, topic, '/external_trigger', 1, indexoffset = -1, bag_from_to=from_to)` ([ref](https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/LrfSensor.py#L34-L35)) in [the "LrfSensor.py" file](https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/LrfSensor.py)? 
What related data do I need in the CSV file (to make a BAG file) related to the "/external_trigger" variable? 

I need to figure out this to artificially make a BAG file. 

----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- 
Here is details:  

I try to run the example command line from [Laser Calibration](https://github.com/ethz-asl/kalibr/wiki/Camera-IMU-LRF-calibration):   
`kalibr_calibrate_imu_camera_laser --cam <cam chain file> --imu <IMU file> --target <April Tag> --imu-models calibrated --bag <BAG file> --reprojection-sigma 0.1 --time-calibration --timeoffset-padding 0.1 --lrf-topic </scan> --q_bl -0.5 0.5 -0.5  0.5 --t_bl 0. 0. 0.` with partially given [dataset](https://github.com/ethz-asl/kalibr/wiki/downloads#sample-datasets) (the [dataset download link](https://drive.google.com/file/d/0B4rISk5dxJScOEhXQ3loMUw1SGM/view?usp=sharing) in instruction page looks not working).

The BAG file isn't supplied, so I am trying to fabricate a BAG file from CSV and image file by myself using `kalibr_bagcreater –-folder <data folder> –-output <BAG file>` [ref](https://github.com/ethz-asl/kalibr/wiki/bag-format).   

What headers do I need in the CSV file?

Currently, I have these header in the CSV file:
&nbsp;&nbsp; timestamp [ns]
&nbsp;&nbsp; time_increment [ns s^-1]
&nbsp;&nbsp; angle [rad]
&nbsp;&nbsp; angle_increment [rad s^-1]
&nbsp;&nbsp; ranges [m]
&nbsp;&nbsp; intensities [w cm^-2],
&nbsp;&nbsp; external_trigger [ns]

I couldn't run the command line because and error of empty array at 
`for (topic, data, stamp) in rosbag.Bag(bagfile).read_messages(topics={triggertopic}):` ([ref](https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_common/ScanDatasetReader.py#L140) in [the "ScanDatasetReader.py" file](https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_common/ScanDatasetReader.py) called from `reader = kc.BagTriggeredScanDatasetReader(bagfile, topic, '/external_trigger', 1, indexoffset = -1, bag_from_to=from_to)` ([ref](https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/LrfSensor.py#L34-L35)) in [the "LrfSensor.py" file](https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/LrfSensor.py). 

Please, give me some guides. 

If the CSV file is custom in various case, it will be great if I could download any example BAG file.

goldbattle, 2022-05-10:
I think this experimental feature has been lost to time, so you will likely need to reverse engineer it (see issue https://github.com/ethz-asl/kalibr/issues/504). The branch hasn't been touched in over 6 years. I am surprised you are able to even build it to run.
https://github.com/ethz-asl/kalibr/tree/experimental/lrf-calibration

I believe you need a ROS topic recorded into your bag file of your 2D laser scanner. I don't think you use a CSV file at all here and all data should be inside the bag from a quick glance at it.

cloudlakecho, 2022-05-10:
@goldbattle Thanks for information and suggestion. 

Regarding the issue 504, I already asked the writer of the issue. No update in there. 

Yes, I am trying to fabricate a (fake) BAG file. As you suggested, I will use the information from [the sensor_msgs/LaserScan](http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/LaserScan.html).

cloudlakecho, 2022-05-14:
@goldbattle I came with one more question of specific (ROS) sensor message (TriggeredScan). 

As you suggested, I followed [the ROS Message web page](http://docs.ros.org/en/api/sensor_msgs/html/msg/LaserScan.html) and I am able to make my own the LaserScan ROS sensor message and convert to the BAG file. (CSV --> ROS Sensor Message --> BAG).

[The LRF calibration](https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera_laser) actually required three different sensor messages: 
&nbsp;&nbsp; LaserScan
&nbsp;&nbsp; Image 
&nbsp;&nbsp; TriggeredScanData

I made top two sensor messages, but the last one I am not able to. Here is details:
 
The TriggerdScanData sensor message ([ref](https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_common/ScanDatasetReader.py#L130)). This senor message format (I am sure if it is ROS format) included:
&nbsp;&nbsp; topic
&nbsp;&nbsp; data (trigger_id, angle_min, angle_increment, ranges, intensities) ([ref](https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_common/ScanDatasetReader.py#L175-L176))
&nbsp;&nbsp; stamp (time stamp)  

I could find all of them at LaserScan sensor message format except the "trigger_id" ([ref](https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_common/ScanDatasetReader.py#L141)). I went to through [entire sensor message formats](https://wiki.ros.org/sensor_msgs), but none of them has the "trigger_id".

It means I have broken pipeline (CSV - x -> ROS Sensor Message --> BAG).

Do you guess the TriggerScanData is old ROS sensor message format or ETHZ own? If ETHZ own format, is there any code I could make a (customized) ROS sensor message format? 

----- ----- ----- ----- ----- ----- ----- ----- ----- ----- ----- -----
Regarding how I run the six year old branch code, I made a Docker Image using [the Dockerfile at Stereolabs](https://github.com/stereolabs/kalibr/blob/master/docker/Dockerfile) with the ETHZ branch, lrf-experimental.

goldbattle, 2022-05-16:
I think you can just switch to the non-trigger reader:
https://github.com/ethz-asl/kalibr/blob/experimental/lrf-calibration/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/LrfSensor.py#L33-L35

I believe the trigger is a custom topic that has "events" when the laser
scanner has a valid message that should be used for the calibration.
No clue, I have never run this code, so your best guess is mine.

cloudlakecho, 2022-05-16:
@goldbattle Thanks for advice. I will read through "LrfSenser" code again.

cloudlakecho, 2022-05-17:
@goldbattle Your suggestion of using an non trigger reader works!

I am able to run the laser calibration code with this command line input in fake sensor data:
` kalibr_calibrate_imu_camera_laser --cam /mnt/kalibr/for_calib_laser/cam_april-camchain.yaml --imu /mnt/kalibr/for_calib_laser/imu_adis16448.yaml --target /mnt/kalibr/for_calib_laser/april_6x6.yaml --imu-models calibrated --bag /mnt/kalibr/result/laser_imu.bag --reprojection-sigma 0.1 --time-calibration --timeoffset-padding 0.1 --lrf-topic  /scan_data_updated --q_bl -0.5 0.5 -0.5  0.5 --t_bl 0. 0. 0.`

This is a part of my calibration result:

  Orientation prior camera-imu found as: (T_i_c)
[[ 0.01540094  0.99947646 -0.02845374]
 [-0.99986     0.01558056  0.00610173]
 [ 0.00654186  0.02835579  0.99957649]]
  Gyro bias prior found as: (b_gyro)
[-0.00315449  0.02495663  0.0787223 ]

I also find out that I need laser, IMU and image sensor data in BAG file.

closed 2022-05-17 by cloudlakecho

goldbattle, 2022-05-18:
If you collect a bag that works, feel free to post and I can update the wiki with details.

goldbattle, 2022-05-18:
If you have a google drive, you can share it with my at pgeneva@udel.edu or upload it to some service that I can download it from.

cloudlakecho, 2022-05-18:
@goldbattle Thanks for the suggestion. I followed it. Please, check the e-mail. 

There are two modification to run the `kalibr_calibrate_imu_camera_laser --cam /mnt/kalibr/for_calib_laser/cam_april-camchain.yaml --imu /mnt/kalibr/for_calib_laser/imu_adis16448.yaml --target /mnt/kalibr/for_calib_laser/april_6x6.yaml --imu-models calibrated --bag /mnt/kalibr/result/laser_imu.bag --reprojection-sigma 0.1 --time-calibration --timeoffset-padding 0.1 --lrf-topic /scan_data_updated --q_bl -0.5 0.5 -0.5 0.5 --t_bl 0. 0. 0.` command line. 

First, please comment out all `import cv` in related to "kalibr_calibrate_imu_camera_laser" code. I remember there were around three Python codes had the tool importing line. Except one, the "cv" isn't actually used in the codes. (see the second instruction)

Second, it is relate to the "cv" tool as well.  One of code will use "cv". Please, replace with "cv2".

goldbattle, 2022-10-11:
Great, many thanks.
I will try to look at it in the upcoming weeks when I get some free time.
Are you ok with me publicallys sharing the bag if everything works out ok?

cloudlakecho, 2022-10-11:
@goldbattle Yes feel free to share in public. The information in the file is randomly generated.