ethz-asl/kalibr #531: how to figure out the param correctly from imu datasheet

Maxinjun, 2022-05-31:
there is a M-G370 datasheet as blow
[image]
 question1:
according to the 'IMU Noise Model - From the Datasheet of the IMU' wiki, sigmal_g denoted as  velocity random walk or  noise density,
but there are  both two params in the above, which one should I use?
question 2:
bias instability's unit is deg/h int the datasheet, which is different with the wiki, how do I use this param correctly？

goldbattle, 2022-06-02:
The wiki provides a pretty complete summary. I also recommend searching existing issues on this repo for help. Generally, one should use the  an allan deviation plot repo to generate it for your specific IMU.

rmadan2, 2025-02-14:
I think that the wiki should clearly state that Noise and Bias are to be calculated from https://github.com/ori-drs/allan_variance_ros, since if one happens to be unfamiliar with IMU calibration, one can assume Kalibr calculates them too.