ethz-asl/kalibr #532: run " rosrun kalibr kalibr_calibrate_imu_camera  " error ! How to solve?

fengyikawhi, 2022-06-08:
Traceback (most recent call last):
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 221, in writeYaml
    with open(filename, 'w') as outfile:
FileNotFoundError: [Errno 2] No such file or directory: 'camchain-imucam-/home/slam/kalibr_ws/data/imu_april.yaml'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/icc_calibrator.py", line 231, in saveCamChainParametersYaml
    chain.writeYaml(resultFile)
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 224, in writeYaml
    self.raiseError( "Could not write configuration to {0}".format(self.yamlFile) )
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 234, in raiseError
    raise RuntimeError( "{0}{1}".format(header, message) )
RuntimeError: [CameraChainParameters Reader]: Could not write configuration to /home/slam/kalibr_ws/data/cam_april-camchain.yaml

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/slam/kalibr_ws/devel/lib/kalibr/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 245, in <module>
    main()
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 219, in main
    iCal.saveCamChainParametersYaml(yamlFilename)
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/icc_calibrator.py", line 233, in saveCamChainParametersYaml
    raise RuntimeError("ERROR: Could not write parameters to file: {0}\n".format(resultFile))
RuntimeError: ERROR: Could not write parameters to file: camchain-imucam-/home/slam/kalibr_ws/data/imu_april.yaml

goldbattle, 2022-06-08:
Please provide the full command and configuration files you use. From the error, it seems you pointed to an invalid location.

fengyikawhi, 2022-06-09:
> Please provide the full command and configuration files you use. From the error, it seems you pointed to an invalid location.
Appreciate your reply， it seems solve my problem ， thank you very much。

closed 2022-06-09 by goldbattle