ethz-asl/kalibr #537: Singular Matrix issue on camera validation

saching13, 2022-06-25:
I have two Fisheye cameras with 180 Deg FOV. I was trying to do stereo calibration using omni-radtan model but encountered the following issue on validation. Any suggestions on how to overcome this ?

```
➜  kalibr_bags rosrun kalibr kalibr_camera_validator --cam omni-radtan-test-camchain.yaml --target april_6x6_80x80cm.yaml
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.064 [m]
    Spacing 0.0192 [m]
initializing camera geometry
Camera /stereo_inertial_publisher/left/image_raw:
  Camera model: omni
  Focal length: [1002.5692444857015, 1005.4935095836079]
  Principal point: [637.6776275100682, 358.86173277919795]
  Omni xi: 2.0443492843408433
  Distortion model: radtan
  Distortion coefficients: [-0.11604615769371034, 0.45292288562081956, -0.00037304896853662407, -0.0003430209815233101]
initializing camera geometry
Camera /stereo_inertial_publisher/right/image_raw:
  Camera model: omni
  Focal length: [991.1462040077117, 993.9745275717806]
  Principal point: [634.8044972035357, 359.9922330463282]
  Omni xi: 2.034611648180097
  Distortion model: radtan
  Distortion coefficients: [-0.12403769230569024, 0.49585542981612835, -0.0009685380713238949, -0.0005033730597604979]
Traceback (most recent call last):
  File "/home/sachin/Desktop/luxonis/kalibr_workspace/devel/lib/kalibr/kalibr_camera_validator", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/sachin/Desktop/luxonis/kalibr_workspace/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_validator", line 498, in <module>
    chain_validator = CameraChainValidator(camchain, targetConfig)
  File "/home/sachin/Desktop/luxonis/kalibr_workspace/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_validator", line 102, in __init__
    edge["A"][cidx_dest] = self.prepareStereoRectificationMaps(cidx_src, cidx_dest)
  File "/home/sachin/Desktop/luxonis/kalibr_workspace/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_validator", line 330, in prepareStereoRectificationMaps
    c1 = -np.linalg.inv(Poa[:,0:3]) * Poa[:,3]
  File "<__array_function__ internals>", line 180, in inv
  File "/home/sachin/.local/lib/python3.8/site-packages/numpy/linalg/linalg.py", line 552, in inv
    ainv = _umath_linalg.inv(a, signature=signature, extobj=extobj)
  File "/home/sachin/.local/lib/python3.8/site-packages/numpy/linalg/linalg.py", line 89, in _raise_linalgerror_singular
    raise LinAlgError("Singular matrix")
numpy.linalg.LinAlgError: Singular matrix
➜  kalibr_bags 
```

goldbattle, 2022-06-28:
I don't think the validator supports omni directional cameras. It should only support pinhole ones since it relies on opencv for undistortion.

Does this issue help you https://github.com/ethz-asl/kalibr/issues/153?

saching13, 2022-07-06:
Ah got it. Thanks for the reply @goldbattle. And sorry for the delay on follow-up. I was on vacation. 

> Does this issue help you https://github.com/ethz-asl/kalibr/issues/153?

Not really. It talks about how pinhole-radtan and pinhole-equi is supported. But I need omni-radtan undistort to check how well the undistort has happened here. 

Do you have any suggestions on that? 

The end goal for me here is to have a stereo setup using 2 cameras which are 180 Deg fov. While retaining atleast 140 deg FOV.

saching13, 2022-07-06:
[image]
[image]

goldbattle, 2022-08-18:
Can you try this package?
Sorry that I have not added this link to the wiki yet.
https://github.com/ethz-asl/image_undistort#image_undistort_node

closed 2023-05-24 by goldbattle