ethz-asl/kalibr #538: Camera-IMU Extrinsics Calibration results seem off

Kaapeine, 2022-06-29:
I am using Kalibr to calibrate the Camera-IMU system on the Qualcomm RB5. This board comes with the OV9282 tracking camera, which has an FOV of 150 degrees. I selected the Kannala-Brandt camera model (Pinhole-Equi) since VINS also provides support for this model.

The main issue I am facing is that the translation values in the Extrinsics provided by Kalibr don't seem to match our measurements. Kalibr suggests that the IMU is 9cm away from the camera.. The values are actually impossible since the entire width of the board is <9cm. Considering the location of the IMU, the translation should be around 4cm. 

However, VINS Mono seems to work fine with this calibration and the optimized extrinsics that it calculates is similar to the values that are given by Kalibr. What exactly is going on here? It seems like there is something going on with the camera model that I don't fully understand. When I used Kalibr with Intel Realsense D435i, the extrinsics were accurate and matched our measurements. 

Here are the steps I took to get these values: 

The following Intrinsics were first obtained. 
```
Calibration results 
====================
Camera-system parameters:
        cam0 (/tracking/image):
         type: <class 'aslam_cv.libaslam_cv_python.DistortedPinholeCameraGeo
metry'>
         distortion: [-0.30052885  0.09126256 -0.00351683 -0.0001841 ] +- [0
.00223441 0.00179199 0.00049859 0.00018655]
         projection: [348.09703409 347.59828813 322.42052705 247.61441954] +
- [1.06914302 1.02405129 0.61522308 0.92603203]
         reprojection error: [-0.000000, -0.000000] +- [0.181675, 0.194522]

Target configuration
====================

  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.088 [m]
    Spacing 0.0264 [m]
```
Next, I obtained IMU noise values from 'imu_utils' and used Kalibr's IMU Camera calibration tool. The results of that are as follows:

```
Calibration results
===================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 0.225606395972, median 0.195595131397, std: 0.173208939403
Gyroscope error (imu0):        mean 0.28782716047, median 0.223148694019, std: 0.254501730529
Accelerometer error (imu0):    mean 0.186703898464, median 0.151224894181, std: 0.144587103885

Residuals
----------------------------
Reprojection error (cam0) [px]:     mean 0.225606395972, median 0.195595131397, std: 0.173208939403
Gyroscope error (imu0) [rad/s]:     mean 0.00564605043677, median 0.00437731025548, std: 0.00499233499878
Accelerometer error (imu0) [m/s^2]: mean 0.0656878231512, median 0.0532052848748, std: 0.0508699185613

Transformation (cam0):
-----------------------
T_ci:  (imu0 to cam0): 
[[-0.01378727 -0.99990409  0.001316    0.00331511]
 [-0.71069107  0.0088736  -0.70344827  0.05811393]
 [ 0.70336912 -0.0106339  -0.71074524 -0.07108811]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam0 to imu0): 
[[-0.01378727 -0.71069107  0.70336912  0.09134794]
 [-0.99990409  0.0088736  -0.0106339   0.00204316]
 [ 0.001316   -0.70344827 -0.71074524 -0.00964975]
 [ 0.          0.          0.          1.        ]]

timeshift cam0 to imu0: [s] (t_imu = t_cam + shift)
-0.018244566668735262

Gravity vector in target coords: [m/s^2]
[ 0.00259696 -9.73665006 -1.16878646]

Calibration configuration
=========================

cam0
-----
  Camera model: pinhole
  Focal length: [348.09703408814636, 347.59828812563086]
  Principal point: [322.42052704710034, 247.61441954109114]
  Distortion model: radtan
  Distortion coefficients: [-0.3005288538522166, 0.09126256100932136, -0.00351683153267163, -0.00018410406353821423]
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.088 [m]
    Spacing 0.0264 [m]

IMU configuration
=================

IMU0:
 ----------------------------
  Model: calibrated
  Update rate: 485.0
  Accelerometer:
    Noise density: 0.0159757264024 
    Noise density (discrete): 0.351828878194 
    Random walk: 0.000503170682535
  Gyroscope:
    Noise density: 0.000890721860106
    Noise density (discrete): 0.0196161141553 
    Random walk: 1.18240957641e-05
  T_i_b
    [[1. 0. 0. 0.]
     [0. 1. 0. 0.]
     [0. 0. 1. 0.]
     [0. 0. 0. 1.]]
  time offset with respect to IMU0: 0.0 [s]
```
It suggests that the IMU is ~9cm away from the Camera, in the second transformation matrix. What could be the problem here? Another issue thread here with the same problem suggested there might be time synchronisation issues.

closed 2022-07-04 by Kaapeine

SongkunChen, 2022-08-11:
Hi Kaapeine, have you found the reason of the 9cm extrinsics ?
I met the same problem here with RealsenseL515, which kalibr shows imu about 15cm away from  camera...

Kaapeine, 2022-08-11:
My bad, I should've closed the issue with the solution. It turned out to be something very silly. We have multiple IMUs and the IMU driver was accessing data from one while we assumed it was another. The extrinsics match for the other IMU. Since you work with a realsense camera, I don't think this is your issue. Hope you find an answer!