ethz-asl/kalibr #546: Compile errors on "incremental_calibration"

Cristian-wp, 2022-07-21:
Hello,
I am trying to compile this module but I get this errors:
`Errors     << incremental_calibration:make /home/ctrazzi/imu2cam_ws/logs/incremental_calibration/build.make.001.log
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp: In member function ‘double aslam::calibration::LinearSolver::getNumFlops() const’:
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp:212:23: error: ‘const cholmod_common {aka const struct cholmod_common_struct}’ has no member named ‘SPQR_flopcount_bound’
       return _cholmod.SPQR_flopcount_bound;
                       ^~~~~~~~~~~~~~~~~~~~
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp: In member function ‘double aslam::calibration::LinearSolver::getSymbolicFactorizationTime() const’:
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp:225:25: error: ‘const cholmod_common {aka const struct cholmod_common_struct}’ has no member named ‘SPQR_analyze_time’
         return _cholmod.SPQR_analyze_time;
                         ^~~~~~~~~~~~~~~~~
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp: In member function ‘double aslam::calibration::LinearSolver::getNumericFactorizationTime() const’:
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp:232:25: error: ‘const cholmod_common {aka const struct cholmod_common_struct}’ has no member named ‘SPQR_analyze_time’
         return _cholmod.SPQR_analyze_time;
                         ^~~~~~~~~~~~~~~~~
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp: In member function ‘virtual void aslam::calibration::LinearSolver::initMatrixStructureImplementation(const std::vector<aslam::backend::DesignVariable*>&, const std::vector<aslam::backend::ErrorTerm*>&, bool)’:
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp:294:9: warning: unused parameter ‘useDiagonalConditioner’ [-Wunused-parameter]
         useDiagonalConditioner) {
         ^~~~~~~~~~~~~~~~~~~~~~
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp: In member function ‘void aslam::calibration::LinearSolver::solve(cholmod_sparse*, cholmod_dense*, std::ptrdiff_t, Eigen::VectorXd&)’:
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp:332:18: error: ‘cholmod_common {aka struct cholmod_common_struct}’ has no member named ‘SPQR_analyze_time’
         _cholmod.SPQR_analyze_time = t3 - t2;
                  ^~~~~~~~~~~~~~~~~
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp:347:16: error: ‘cholmod_common {aka struct cholmod_common_struct}’ has no member named ‘SPQR_analyze_time’
       _cholmod.SPQR_analyze_time = t3 - t2;
                ^~~~~~~~~~~~~~~~~
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp: In member function ‘void aslam::calibration::LinearSolver::analyzeMarginal(cholmod_sparse*, std::ptrdiff_t)’:
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp:479:18: error: ‘cholmod_common {aka struct cholmod_common_struct}’ has no member named ‘SPQR_analyze_time’
         _cholmod.SPQR_analyze_time = t3 - t2;
                  ^~~~~~~~~~~~~~~~~
/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/core/LinearSolver.cpp:492:16: error: ‘cholmod_common {aka struct cholmod_common_struct}’ has no member named ‘SPQR_analyze_time’
       _cholmod.SPQR_analyze_time = t3 - t2;
                ^~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/incremental_calibration.dir/src/core/LinearSolver.cpp.o] Error 1
make[1]: *** [CMakeFiles/incremental_calibration.dir/all] Error 2
make: *** [all] Error 2
cd /home/ctrazzi/imu2cam_ws/build/incremental_calibration; catkin build --get-env incremental_calibration | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
`

I have check the library that contain the definition of SPQR_flopcount_bound and SPQR_analyze_time (/usr/local/include/cholmod_core.h) and there is no definition inside.
Suitsparse is the latest version and I have already try to remove and install it with apt-get but nothing change...
Can someone help me with this problem please?

goldbattle, 2022-07-21:
Please provide your operation system information and the version of suitesparse you have installed.

Cristian-wp, 2022-07-22:
Hello @goldbattle , thanks for your reply.
I am workin on Ubuntu 18 with ROS Melodic and to make Kalibr work I have follow the instruction on the wiky page. 

I have installed SuiteSparse with apt-get (`sudo apt-get install libsuitesparse-dev -y`) and is the latest version, this is the output of the installation:

`libsuitesparse-dev is already the newest version (1:5.1.2-2)`

I have even try the "hard way" and copy and paste the library from github master branch (same for the .a after compile them with make).
Now the compilation with catkin_build goes well with 0 errors, but when I launch the calibration with:

`rosrun kalibr kalibr_calibrate_imu_camera --bag 2022-07-20-11-20-56.bag --cam camchain.yaml  --imu imu.yaml --target aprilgrid.yaml`

I get the following output:

`rosrun kalibr kalibr_calibrate_imu_camera --bag 2022-07-20-11-20-56.bag --cam camchain.yaml  --imu imu.yaml --target aprilgrid.yaml 
importing libraries
Traceback (most recent call last):
  File "/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 4, in <module>
    import kalibr_common as kc
  File "/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/__init__.py", line 3, in <module>
    from .ConfigReader import *
  File "/home/ctrazzi/imu2cam_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 9, in <module>
    import aslam_cv_backend as cvb
  File "/home/ctrazzi/imu2cam_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 6, in <module>
  File "/home/ctrazzi/imu2cam_ws/devel/lib/python2.7/dist-packages/aslam_backend/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 5, in <module>
ImportError: /home/ctrazzi/imu2cam_ws/devel/lib/libaslam_backend.so: undefined symbol: omp_get_thread_num
`
The strange thing is that I have try to make this package work on another PC with a fresh Ubuntu 18 and ROS melodic "complete", and the output of catkin_build is the same as the first topic.

Can you please help me?

goldbattle, 2022-07-22:
>  copy and paste the library from github master branch

You should just directly clone the repository into a ROS workspace.
Not sure what you mean by this.

Make sure you are sourcing your workspace after you build it.
Are you able to run any other commands successfully?
Might still be an issue of linking, can you try running ` ldd devel/lib/libaslam_backend.so` to see what libraries are linked.
For example I have (on 20.04):
```
patrick@patrick-ThinkPad-P51:~/workspace/catkin_ws_kalibr$ ldd devel/lib/libaslam_backend.so
	linux-vdso.so.1 (0x00007fff5e165000)
	libboost_thread.so.1.71.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0 (0x00007f92c3e2f000)
	libcholmod.so.3 => /usr/lib/x86_64-linux-gnu/libcholmod.so.3 (0x00007f92c3d4c000)
	libspqr.so.2 => /usr/lib/x86_64-linux-gnu/libspqr.so.2 (0x00007f92c3d1e000)
	libopenblas.so.0 => /usr/lib/x86_64-linux-gnu/libopenblas.so.0 (0x00007f92c1b91000)
	libmetis.so => /usr/local/lib/libmetis.so (0x00007f92c191e000)
	libsparse_block_matrix.so => /home/patrick/workspace/catkin_ws_kalibr/devel/.private/sparse_block_matrix/lib/libsparse_block_matrix.so (0x00007f92c1913000)
	libsm_timing.so => /home/patrick/workspace/catkin_ws_kalibr/devel/.private/sm_timing/lib/libsm_timing.so (0x00007f92c18f1000)
	libsm_logging.so => /home/patrick/workspace/catkin_ws_kalibr/devel/.private/sm_logging/lib/libsm_logging.so (0x00007f92c18ae000)
	libsm_property_tree.so => /home/patrick/workspace/catkin_ws_kalibr/devel/.private/sm_property_tree/lib/libsm_property_tree.so (0x00007f92c1858000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f92c1833000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f92c1651000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f92c1502000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f92c14e7000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f92c12f5000)
	libamd.so.2 => /usr/lib/x86_64-linux-gnu/libamd.so.2 (0x00007f92c12ea000)
	libcolamd.so.2 => /usr/lib/x86_64-linux-gnu/libcolamd.so.2 (0x00007f92c12e1000)
	libsuitesparseconfig.so.5 => /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so.5 (0x00007f92c12da000)
	libccolamd.so.2 => /usr/lib/x86_64-linux-gnu/libccolamd.so.2 (0x00007f92c12cd000)
	libcamd.so.2 => /usr/lib/x86_64-linux-gnu/libcamd.so.2 (0x00007f92c12c1000)
	libmetis.so.5 => /usr/lib/x86_64-linux-gnu/libmetis.so.5 (0x00007f92c1053000)
	liblapack.so.3 => /usr/lib/x86_64-linux-gnu/liblapack.so.3 (0x00007f92bf03f000)
	libblas.so.3 => /usr/lib/x86_64-linux-gnu/libblas.so.3 (0x00007f92bd640000)
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f92bd5fc000)
	libgfortran.so.5 => /usr/lib/x86_64-linux-gnu/libgfortran.so.5 (0x00007f92bd334000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f92c3fc7000)
	libboost_regex.so.1.71.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.71.0 (0x00007f92bd234000)
	libboost_filesystem.so.1.71.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007f92bd216000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f92bd20e000)
	libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007f92bd1c4000)
	libicui18n.so.66 => /usr/lib/x86_64-linux-gnu/libicui18n.so.66 (0x00007f92bcec5000)
	libicuuc.so.66 => /usr/lib/x86_64-linux-gnu/libicuuc.so.66 (0x00007f92bccdf000)
	libicudata.so.66 => /usr/lib/x86_64-linux-gnu/libicudata.so.66 (0x00007f92bb21e000)

```

In the worst case I recommend you use the docker.

Cristian-wp, 2022-07-22:
Thanks for your reply.
With copy and paste I mean that I have try to reinstall libsuitesparse with apt-get and nothing change. So I clone from github libsuitesparse (master branch) in the ros workspace for kalibr, but even in this case I get the same first error.
So I check the precence of the function SPQR_analyze_time and SPQR_flopcount_bound in the library on my pc, but in the file .h there are not the functions. So I have copy and paste manually every compiled library of libsuitesparse in the correct paths of my pc. After that I get the error in my second thread.

I can not run any of the package...but the strange thing is that I have already try to reinstall all in a new ubuntu 18 installation (following kalibr wiky for the dependency) and I get the same errors of my first topic.

goldbattle, 2022-07-26:
Can you try running `ldd devel/lib/libaslam_backend.so` to see what libraries are linked? Otherwise I am not sure what else to debug here to help you.

fushi219, 2022-07-29:
Hi, i got exactly the same error as the first ones.
My system is Ubuntu16.04, with ROS kinetic fully installed, and suitsparse library in 1:4.4.6-1.

My libaslam_backend.so is linked as follows:

> 	linux-vdso.so.1 =>  (0x00007fff4f5b1000)
	libboost_system.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0 (0x00007f63cccdb000)
	libboost_thread.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0 (0x00007f63ccab5000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f63cc898000)
	liblapack.so.3 => /usr/lib/liblapack.so.3 (0x00007f63cc0b4000)
	libf77blas.so.3 => /usr/lib/libf77blas.so.3 (0x00007f63cbe94000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f63cbc8c000)
	libsparse_block_matrix.so => /home/fushi/kalibr_ws/devel/lib/libsparse_block_matrix.so (0x00007f63cba4d000)
	libsm_timing.so => /home/fushi/kalibr_ws/devel/lib/libsm_timing.so (0x00007f63cb71d000)
	libsm_logging.so => /home/fushi/kalibr_ws/devel/lib/libsm_logging.so (0x00007f63cb4cb000)
	libsm_property_tree.so => /home/fushi/kalibr_ws/devel/lib/libsm_property_tree.so (0x00007f63cb217000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f63cae95000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f63cab8c000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f63ca976000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f63ca5ac000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f63cd86a000)
	libblas.so.3 => /usr/lib/libblas.so.3 (0x00007f63ca36e000)
	libatlas.so.3 => /usr/lib/libatlas.so.3 (0x00007f63c9dd0000)
	libgfortran.so.3 => /usr/lib/x86_64-linux-gnu/libgfortran.so.3 (0x00007f63c9aa5000)
	libcblas.so.3 => /usr/lib/libcblas.so.3 (0x00007f63c9883000)
	libboost_regex.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.58.0 (0x00007f63c957b000)
	libboost_filesystem.so.1.58.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0 (0x00007f63c9363000)
	libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007f63c9124000)
	libicui18n.so.55 => /usr/lib/x86_64-linux-gnu/libicui18n.so.55 (0x00007f63c8cc2000)
	libicuuc.so.55 => /usr/lib/x86_64-linux-gnu/libicuuc.so.55 (0x00007f63c892e000)
	libicudata.so.55 => /usr/lib/x86_64-linux-gnu/libicudata.so.55 (0x00007f63c6e77000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f63c6c73000)

Thanks a lot!

fushi219, 2022-07-30:
I remove all files in build and devel folders, recompile the whole project, the problem then solved, wish this could help you? @Cristian-wp

Cristian-wp, 2022-08-12:
Hi, thanks for the suggestions.
I am sorry, but your suggestions werent work so I decided to reinstall a clean version of ubuntu and the problem disappear.

closed 2022-08-12 by goldbattle