ethz-asl/kalibr #549: kalibr_calibrate_cameras command not found

zhonghonglong, 2022-07-25:
kalibr_calibrate_cameras command not found
setting environment variables stiii doesn't work .
source ~/kalibr_workspace/devel/setup.bash

is there anything i can do?

goldbattle, 2022-07-25:
Have you followed the instructions on the wiki?
https://github.com/ethz-asl/kalibr/wiki/installation

Please provide more details about your setup and what you are doing.

zhonghonglong, 2022-07-26:
Problem solved Thank you!

zhonghonglong, 2022-07-26:
> 您是否按照 wiki 上的说明进行操作？ https://github.com/ethz-asl/kalibr/wiki/installation
> 
> 请提供有关您的设置和您正在做什么的更多详细信息。
Sorry,I missed a step .Problem solved Thank you very much!

vsaketharam, 2022-08-02:
I am having similar problem. I followed all steps in the installation instruction from Building Kalibr from Source but I am still getting 'kalibr_calibrate_cameras command not found'. Is there anything else that might be causing this problem?

hszyq1234, 2023-02-28:
> 

You should run "source /kalibr_workspace/devel/setup.bash". Check the path of kalibr_workspace.

Z-star1, 2023-03-07:
i have the same problem, we have run "source /kalibr_workspace/devel/setup.bash"

BIT-DYN, 2023-03-27:
As an alternative, you can run "rosrun kalibr ***".

victorsillerico, 2024-06-10:
> As an alternative, you can run "rosrun kalibr ***".

It worked for me! thanks!