ethz-asl/kalibr #550: ERROR: invalid command 'numpy_eigen'

Abonaventure, 2022-07-28:
ERROR: invalid command 'numpy_eigen'
CMake Error at /opt/ros/noetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):

  execute_process(/mnt/d/ROS_DEV/kalibr_ws/build/numpy_eigen/catkin_generated/env_cached.sh
  "/usr/bin/python3"
  "/opt/ros/noetic/share/catkin/cmake/interrogate_setup_dot_py.py"
  "numpy_eigen"
  "/mnt/d/ROS_DEV/kalibr_ws/src/kalibr/Schweizer-Messer/numpy_eigen/setup.py"
  "/mnt/d/ROS_DEV/kalibr_ws/build/numpy_eigen/catkin_generated/setup_py_interrogation.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkin_python_setup.cmake:46 (safe_execute_process)
  CMakeLists.txt:10 (catkin_python_setup)

goldbattle, 2022-07-28:
Please provide more details, thanks.

Abonaventure, 2022-07-29:
> Please provide more details, thanks.

I compile the kalibr source codes under the following environments:wsl2(ubuntu20.04), other libs follow your Installation web "https://github.com/ethz-asl/kalibr/wiki/installation". My compile command is "catkin build -DCMAKE_BUILD_TYPE=Release -j32 2>&1 | tee kalibr-log",and the kalibr-log is as following:

--------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /mnt/d/ROS_DEV/kalibr_ws
--------------------------------------------------------------------------------
Build Space:        [exists] /mnt/d/ROS_DEV/kalibr_ws/build
Devel Space:        [exists] /mnt/d/ROS_DEV/kalibr_ws/devel
Install Space:      [unused] /mnt/d/ROS_DEV/kalibr_ws/install
Log Space:         [missing] /mnt/d/ROS_DEV/kalibr_ws/logs
Source Space:       [exists] /mnt/d/ROS_DEV/kalibr_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        -j32
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
--------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------
[build] Found 36 packages in 0.0 seconds.                                      
[build] Updating package table.                                                
Starting >>> aslam_time                                                        
Starting >>> catkin_simple                                                     
Starting >>> ethz_apriltag2                                                    
Starting >>> python_module                                                     
Starting >>> sm_common                                                         
Starting >>> sm_logging                                                        
Starting >>> sm_random                                                         
[build 0.8 s] [0/36 complete] [7/32 jobs] [4 queued] [ethz_apriltag2:loaden... 
[build 0.9 s] [0/36 complete] [7/32 jobs] [3 queued] [sm_common:cmake - 0.2... 
[build 1.0 s] [0/36 complete] [7/32 jobs] [1 queued] [sm_common:cmake - 0.3... 
[build 1.1 s] [0/36 complete] [7/32 jobs] [1 queued] [sm_common:cmake - 0.4... 
[build 1.2 s] [0/36 complete] [7/32 jobs] [1 queued] [sm_common:cmake - 0.5... 
[build 1.3 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 0.6... 
[build 1.4 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 0.7... 
[build 1.5 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 0.8... 
[build 1.6 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 0.9... 
[build 1.7 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 1.0... 
[build 1.8 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 1.1... 
[build 1.9 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 1.2... 
[build 2.0 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 1.3... 
[build 2.1 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 1.4... 
[build 2.2 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 1.5... 
[build 2.3 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 1.6... 
[build 2.4 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 1.7... 
[build 2.5 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 1.8... 
[build 2.6 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 1.9... 
[build 2.7 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 2.0... 
[build 2.8 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 2.1... 
[build 2.9 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 2.2... 
[build 3.0 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 2.3... 
[build 3.1 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 2.4... 
[build 3.2 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 2.5... 
[build 3.3 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 2.6... 
[build 3.4 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 2.7... 
[build 3.5 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 2.8... 
[build 3.6 s] [0/36 complete] [7/32 jobs] [0 queued] [sm_common:cmake - 2.9... 
[... 979 lines cut ...]
                 from [01m[K/mnt/d/ROS_DEV/kalibr_ws/src/kalibr/Schweizer-Messer/sm_boost/include/boost/portable_binary_archive.hpp:31[m[K,
                 from [01m[K/mnt/d/ROS_DEV/kalibr_ws/src/kalibr/Schweizer-Messer/sm_boost/include/boost/portable_binary_iarchive.hpp:39[m[K,
                 from [01m[K/mnt/d/ROS_DEV/kalibr_ws/src/kalibr/Schweizer-Messer/sm_boost/src/portable_binary_iarchive.cpp:18[m[K:
[... cut ...]
[build] Summary: 14 of 36 packages succeeded.                                  
[build] Ignored: None.                                                         
[build] Warnings: 14 packages succeeded with warnings.                         
[build] Abandoned: 21 packages were abandoned.                                 
[build] Failed: 1 packages failed.                                             
[build] Runtime: 49.9 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

Abonaventure, 2022-08-15:
I uninstall wsl and ROS, re-install wsl and ROS, the kalibr source codes compile pass.But I don't know why. The following is the comiling log at the end:
......
[build] Summary: All 36 packages succeeded!
[build]   Ignored:   None.
[build]   Warnings:  None.
[build]   Abandoned: None.
[build]   Failed:    None.
[build] Runtime: 16.4 seconds total.

closed 2022-08-18 by goldbattle

lenardxu, 2022-08-22:
> I uninstall wsl and ROS, re-install wsl and ROS, the kalibr source codes compile pass.But I don't know why. The following is the comiling log at the end: ...... [build] Summary: All 36 packages [...]

Hi, I also met the same problem. By referring to your solution, I reinstalled the ROS noetic. However, the same problem during building still exists. For that, you could check [my post](https://github.com/ethz-asl/kalibr/issues/562) for more details. Thanks anyway and still seeking for help from public!

lenardxu, 2022-08-22:
> Hi, I also met the same problem. By referring to your solution, I reinstalled the ROS noetic. However, the same problem during building still exists. For that, you could check [my [...]

I've solved that problem (possibly specific to my case) on my own. See still [my post](https://github.com/ethz-asl/kalibr/issues/562) for more details.