ethz-asl/kalibr #551: Feature Request: Linting

bagavi, 2022-07-31:
I was wondering if the cpp part of can be linted. This will be very useful to work with modern editors.

Possible option
1. Include a standard `.clang-format` [clang-format.txt](https://github.com/ethz-asl/kalibr/files/9228049/clang-format.txt) at the top folder.
2. Run `find . -regex '.*\.\(cpp\|hpp\|cc\|cxx\)' -exec clang-format -style=file -i {} \;` at the top folderl

PS: If all python code is moved to python3, we could perhaps lint that next :)

goldbattle, 2022-07-31:
This is probably too big of a change since there is a lot of packages that are used in other ASL projects (e.g., [these](https://github.com/ethz-asl?q=aslam&type=all&language=&sort=)). Thus after formating, updating them from upstream repos might be an issue (this has yet to ever been done in many years through...).

Will leave this issue open for any other contributors to chime in on if this is desired.

goldbattle, 2022-08-18:
Closing for now.