ethz-asl/kalibr #552: Large Extrinisic calibration error between D415 cmera image and IMU

ahar, 2022-08-05:
Hello,

We have tried to use the Kalibr package to perform extrinsic calibration of a D415 realsense camera and an IMU. A sample dataset is provided with [this](https://hitechroboticsystemz2-my.sharepoint.com/personal/vikash_kumawat_hitechroboticsystemz_com/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fvikash%5Fkumawat%5Fhitechroboticsystemz%5Fcom%2FDocuments%2Fvikash%2Dxsens%2Dcam%2Ddataset&ga=1) link. It also contains the calibration result we obtained. Upon carefully looking at the results we realize that the translation components contained within `T_ci` reflects a large distance between the IMU and the camera, which does not match our hardware setup. The error is around ten times the original numbers. Based on these observations and our experience using this package the following are the queries I have and prompt response on the same will help me a lot to understand this software better.
Queries:
1. Are the coordinate frames of the camera and the IMU required to be the same? As ROS assumes right-handed coordinate system, I assume that in whatever orientation I keep these two devices, the data collected will be in individual body frames maintaining the right-handed system. Is that right?
2. If my assumptions in (1) are correct, how are the coordinate frames adjusted? I mean either the camera's or the IMU's frame must be considered as the reference right?
3. While going through some of the existing issues I observed that the delay between the camera and the IMU data plays a role and it is suggested to maintain the delay below 20 msec, which is for the data uploaded in the link above. What could be other reasons behind the wrong transformation?

ahar, 2022-08-18:
@goldbattle could you please help resolve my queries?

goldbattle, 2022-08-18:
Sorry, have very minimal throughput this time of the year.
Here are some initial comments.

1. Yes, everything should be right handed. If you have a left handed IMU publisher then the driver will need to be fixed. Personally, I have calibrated the T265 and D455 with no issues with Intel's driver.
2. The result in the yaml is for a specific direction, but the results.txt file has both transforms (in IMU or in CAMERA) in plan english.
3. If your secondary system can handle a time offset, then it is fine to have this offset. But 20ms is very large and you might want to check that your timestamps are good and monotonic.

I would point you in another direction from briefly looking at your results. You need to calibrate your IMU noises and inflate them a bit (2x for white, and 10x for walks).
- https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model#how-to-obtain-the-parameters-for-your-imu
- https://github.com/ori-drs/allan_variance_ros

For example these should be "in bounds"
[image]
[image]

And your reprojection error should be much much lower than 5 pixel.
I wouldn't expect any good performance if it is this large....
[image]

ahar, 2022-08-25:
Hello @goldbattle 
Based on your initial response we looked into `time-offset` from the recorded bag file shared in my first post. I found that the camera data was reporting at `15Hz` instead of at `30 Hz`. In that,  `timeoffset ` exceeded 20ms (approx. `35Hz`). Upon recording the dataset once more while ensuring the camera frame rate of `30Hz` (reference earlier results), the calibration steps were performed again. 

The results obtained for `Xsens-630 IMU` + `D415` can be found [here](https://hitechroboticsystemz2-my.sharepoint.com/:f:/g/personal/vikash_kumawat_hitechroboticsystemz_com/EucCj96PeoBDkRdp6f5UJZcBm6aGB2MtGTrTd5KxPhg7zw?e=HVkgUc) (unscaled noise). The same for `Lord-CV7 IMU`+ `D415` can be found [here](https://hitechroboticsystemz2-my.sharepoint.com/:f:/g/personal/vikash_kumawat_hitechroboticsystemz_com/Eq4AsQk6z6dFlNUaoeFTspsBi7ri0docNpPo1Sp8tNhJOg?e=PMv8Z7). Both the IMUs were intrinsically calibrated beforehand using `Allan-Variance-Ros` package. `Imu.yaml` files for individual IMU containing `random-walk` and `noise-density` can be found in the abovementioned link respectively.

As you suggested, the extrinsic IMU+camera calibration with scaled-up noise parameters are also performed and can be found in [link1](https://hitechroboticsystemz2-my.sharepoint.com/:f:/g/personal/vikash_kumawat_hitechroboticsystemz_com/En6-oRnwVKxGnashGK0P9WsBQSRPuc_4Hpr6AhZtWC3KEw?e=XXZ5T2) (Xsens-630) and [link2](https://hitechroboticsystemz2-my.sharepoint.com/:f:/g/personal/vikash_kumawat_hitechroboticsystemz_com/EoWvZ9ICXNVBpJk20lpzJGQBMCaIvuBwy_J2S0W8gqj5qA?e=OXCM23) (LORD CV7).

**_For **Set-1** (Xsens630Imu with D415 realsense camera):_**

In the calibration [results](https://hitechroboticsystemz2-my.sharepoint.com/:f:/g/personal/vikash_kumawat_hitechroboticsystemz_com/En6-oRnwVKxGnashGK0P9WsBQSRPuc_4Hpr6AhZtWC3KEw?e=XXZ5T2) with scaled-up noise parameters, `time-offset ` appears to be `8.25ms`. Also, the translation components in `imu0 to cam0` Extrinsic transformation matrix has now come down to sub-cm level except for the first element of the translation vector. We expect it to be around `1cm` whereas the calibration result shows approx. `5.2 cm`.
***I would like to know your observations on it (screenshot attached). 
[image]

With unscaled noise parameters [link](https://hitechroboticsystemz2-my.sharepoint.com/:f:/g/personal/vikash_kumawat_hitechroboticsystemz_com/EucCj96PeoBDkRdp6f5UJZcBm6aGB2MtGTrTd5KxPhg7zw?e=HVkgUc) of the IMUs the optimization failed by printing **that matrix is not positive
definite**. (screenshot below) :      [image]

Here i have few questions ::-
1)
2)

**_For **Set-2** (Lord-CV7 Imu with D415 realsense camera):_**

For Lord-IMU and Camera as well , Calibration was performed with `Scaled and Unscaled Imu.yaml file`. in the case of Lord-IMU  Calibration was `successfully `completed for `both yaml files.` Results we got was also almost good.  Here I am attaching Results of both Calibration.

LORD-IMU-UNSCALED Data- : [ https://hitechroboticsystemz2-my.sharepoint.com/:f:/g/personal/vikash_kumawat_hitechroboticsystemz_com/Eq4AsQk6z6dFlNUaoeFTspsBi7ri0docNpPo1Sp8tNhJOg?e=PMv8Z7](url)
these results shows time shift is around `5.4ms.` Extrinsic transformation matrix between imu and camera is almost nearby but again same one translation value we are expecting aroung `1cm` and it gives around `4.4 cm` (same issue with `Xsens630`) . you can see below or can check results attached above:
[image]

LORD-SCALED-DATA:- [https://hitechroboticsystemz2-my.sharepoint.com/:f:/g/personal/vikash_kumawat_hitechroboticsystemz_com/EoWvZ9ICXNVBpJk20lpzJGQBMCaIvuBwy_J2S0W8gqj5qA?e=OXCM23](url)

Similarly here we also got one wrong  translation value (for same axis) . it should be `1cm` but getting `5.1cm.`
[image]

#

ahar, 2022-09-07:
@goldbattle any views on my above query?

goldbattle, 2022-09-14:
The failure is likely due to the noises being too small causing numerical issues, but I am not sure.

closed 2023-07-20 by goldbattle