ethz-asl/kalibr #553: RuntimeError: ERROR: Could not write parameters to file: camchain-imucam-/home/slam/record_rosbag_ws/data20220804/cam_imu_RGB4.yaml

fengyikawhi, 2022-08-05:
Traceback (most recent call last):
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/icc_calibrator.py", line 231, in saveCamChainParametersYaml
    chain.writeYaml(resultFile)
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 221, in writeYaml
    os.mkdir(filename)
FileNotFoundError: [Errno 2] No such file or directory: 'camchain-imucam-/home/slam/record_rosbag_ws/data20220804/cam_imu_RGB4.yaml'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/slam/kalibr_ws/devel/lib/kalibr/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 245, in <module>
    main()
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 219, in main
    iCal.saveCamChainParametersYaml(yamlFilename)
  File "/home/slam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/icc_calibrator.py", line 233, in saveCamChainParametersYaml
    raise RuntimeError("ERROR: Could not write parameters to file: {0}\n".format(resultFile))
RuntimeError: ERROR: Could not write parameters to file: camchain-imucam-/home/slam/record_rosbag_ws/data20220804/cam_imu_RGB4.yaml

Your help is much appreciated.

Best.

goldbattle, 2022-08-05:
What is the exact command you are using to run the program with?
Can you try to use a relative path?
Seems this could be an issue with how to parse the bag file path.

fengyikawhi, 2022-08-08:
> What is the exact command you are using to run the program with? Can you try to use a relative path? Seems this could be an issue with how to parse the bag file path.

I tried command:  "rosrun kalibr kalibr_calibrate_imu_camera --bag ./data/20220802/cam_imu_RGB4.bag --target ./data/20220802/checkerboard.yaml --cam ./data/20220802/camchain.yaml --imu ./data/20220802/imu.yaml"
But all most the same problems:
FileNotFoundError: [Errno 2] No such file or directory: 'camchain-imucam-./data/20220802/cam_imu_RGB4.yaml'
RuntimeError: ERROR: Could not write parameters to file: camchain-imucam-./data/20220802/cam_imu_RGB4.yaml

Any ideas to solve this?

fengyikawhi, 2022-08-09:
> The export order is `bagname + -camchain-imucam.yaml`. Can you please make sure the code you compiled is up to date with the current master? https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera#L220

Appreciate for your patient reply！

closed 2022-08-18 by goldbattle