ethz-asl/kalibr #559: [calibrate_imu_camera] with too big camera reprojection error!

SongkunChen, 2022-08-16:
Hi friends:

   I  try to calibrate a MPU6050 with a depth camera; I got the good state for the camera calibration alone, with reprojection erros to be about 0.4-0.5 pixel ( in the moving state); 
   But when I try to calibrate the camera and IMU as a whole, the camera reprojection error is too big ( to nearly 3-4 pixels). I try to increase the noise property of the IMU (with 2x noise density and 10x randwalk),  but the result does not get any improvent. 
 
Here is the calibration result. Any suggestions for the further modification?
Sincere thanks in advance!
[test_420_kalibr-report-cam.pdf](https://github.com/ethz-asl/kalibr/files/9346843/test_420_kalibr-report-cam.pdf)
[test_420_kalibr-report-imucam.pdf](https://github.com/ethz-asl/kalibr/files/9346258/test_420_kalibr-report-imucam.pdf)

SongkunChen, 2022-08-17:
Update:
   1. I finally found that , my camera model was not well chosen. Refer to the following repo, I managed to validate the camera model, and choose a better one for my camera, which shows a better result. Hope the information will help someone else.
   https://github.com/ethz-asl/image_undistort#loading-camera-information-from-ros-parameters

2. I am a little confused by the following optimization options. Could some share some light on it ,so that I can get better simulation results between body-estimation and measured data?
    iCal.buildProblem(splineOrder=6, 
                      poseKnotsPerSecond=100, 
                      biasKnotsPerSecond=100, 
                      doPoseMotionError=False,
                      doBiasMotionError=True,
                      blakeZisserCam=-1,
                      huberAccel=-1,
                      huberGyro=-1,
                      noTimeCalibration=parsed.no_time,
                      noChainExtrinsics=(not parsed.recompute_chain_extrinsics),
                      maxIterations=parsed.max_iter,
                      timeOffsetPadding=parsed.timeoffset_padding,
                      verbose = parsed.verbose)
[image]

SongkunChen, 2022-09-22:
Issue Resolved;
Due to the wrong set of my own camera.

closed 2022-09-22 by SongkunChen

foemre, 2023-06-12:
> Issue Resolved; Due to the wrong set of my own camera.

What did you change in your final calibration? What was wrong in your previous setup?