ethz-asl/kalibr #561: Strange Reprojection Error Behavior

yooooki, 2022-08-18:
Hi friends:

I'm trying to calibrate a FLIR BFS-U3-16S2C camera. The reprojection in the calibration report looks weird:
[image]
The result was
```
Calibration results 
====================
Camera-system parameters:
	cam0 (/blackfly/cam0/image_raw):
	 type: <class 'aslam_cv.libaslam_cv_python.DistortedPinholeCameraGeometry'>
	 distortion: [-0.13501326  0.02032476 -0.00427919  0.00106607] +- [0.00426444 0.001329   0.0004348  0.0002439 ]
	 projection: [737.58096631 716.35258317 671.88113768 567.86336969] +- [12.67128558 12.26143029  2.45023513  4.33137152]
	 reprojection error: [0.000001, 0.000001] +- [5.273541, 5.111451]
```
Other camera models failed to complete the optimization or initialization.

I suspect this is due to the lens I used was not suitable for computer vision and may not physically fit these camera models, but I can't find any valuable information about my situation.
So could this just be a problem with my lens, or other mistakes I made during the recording process?

yooooki, 2022-09-02:
It turns out that a silly misunderstanding of parameter `tagSpacing` in the target file led to this problem.

closed 2022-09-02 by yooooki

elisaaferraraa, 2024-11-13:
> It turns out that a silly misunderstanding of parameter `tagSpacing` in the target file led to this problem.

Hey @yooooki , could you clarify your understanding of `tagSpacing`? From my understanding it should be the ratio: (white space between tags)/(size of one single tag) . Given this, if apriltag proportions are unchanged, `tagSpacing` should be constantly equal to 0.3, do you agree ?