ethz-asl/kalibr #564: calibration method

dirux0101, 2022-08-29:
Hello,

I calibrated realsense D455 using Kalibr, but the value did not come out well. I'm wondering if the method was wrong during the correction process. The topics of imu, camera, and imu/camera were saved and corrected in each rosbag. In this process, Aprilgrid(A0) was used, and when saving the rosbag, the camera was moved around and the bag file was saved. Is my correction process correct? Or is there another way to move the camera while saving the rosbag?

goldbattle, 2022-09-14:
My lab has been able to calibrate the D455, you might want to ensure that you are "combining" the IMU topics as the realsense driver likely is only publishing the acceleration and gyroscope as different topics.
https://github.com/ethz-asl/kalibr/issues/458

dirux0101, 2022-09-25:
Sorry for late. Can you explain more precisely?

My camera topic is publish like this:
[image]

dirux0101, 2022-09-25:
My command is : roslaunch realsense2_camera rs_camera.launch filters:=pointcloud unite_imu_method:="linear_interpolation" enable_gyro:=true enable_accel:=true

As a result of checking, it seems that it is correct to come out combined. Anything else to keep in mind while performing the calibration?

Thank you

goldbattle, 2022-10-21:
No, that should be it. [This repo](https://github.com/rpng/ar_table_dataset) and some bags / scripts / results from a d455. Hope this helps you.

closed 2022-11-24 by goldbattle