ethz-asl/kalibr #566: Could NOT find Boost (missing: python38) (found version "1.67.0")

zoldaten, 2022-09-07:
hi!
i am building from source  on raspberry pi raspbian buster and crashing:
```
-- +++ processing catkin package: 'numpy_eigen'
-- ==> add_subdirectory(kalibr/Schweizer-Messer/numpy_eigen)
-- Found Boost: /usr/include (found version "1.67.0") found components: system 
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find Boost (missing: python38) (found version "1.67.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:2179 (find_package_handle_standard_args)
  kalibr/Schweizer-Messer/python_module/cmake/add_python_export_library.cmake:91 (find_package)
  kalibr/Schweizer-Messer/numpy_eigen/CMakeLists.txt:38 (add_python_export_library)

-- Configuring incomplete, errors occurred!
See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```
i tried to add to kalibr/Schweizer-Messer/numpy_eigen/CMakeLists.txt:

```
find_package(PythonLibs 3.6 REQUIRED)
# Extract major/minor python version
string(REPLACE "." ";" VERSION_LIST ${PYTHONLIBS_VERSION_STRING})
list(GET VERSION_LIST 0 PYTHONLIBS_VERSION_MAJOR)                                                                                                                             
list(GET VERSION_LIST 1 PYTHONLIBS_VERSION_MINOR)
find_package(Boost COMPONENTS python${PYTHONLIBS_VERSION_MAJOR}${PYTHONLIBS_VERSION_MINOR} REQUIRED)
```
but no result.
and it crashes only on numpy_eigen.

could you help a bit ?

goldbattle, 2022-09-08:
You might want to try to upgrade your boost installation.

zoldaten, 2022-09-08:
Ok.
As `sudo apt-get install libboost-all-dev` said that i have a latest 1.67 version of boost 
I built boost-1.80 from sorce:
```
wget https://boostorg.jfrog.io/artifactory/main/release/1.80.0/source/boost_1_80_0.tar.gz
tar xvf boost_1_80_0.tar.gz 
cd cd boost_1_80_0
./bootstrap.sh
sudo ./b2
```
It said that:
```
The Boost C++ Libraries were successfully built!
The following directory should be added to compiler include paths:
    /home/pi/boost_1_80_0
The following directory should be added to linker library paths:
    /home/pi/boost_1_80_0/stage/lib
```

After that i tried to rebuild kalibr but got error changed:
```
-- +++ processing catkin package: 'numpy_eigen'
-- ==> add_subdirectory(kalibr/Schweizer-Messer/numpy_eigen)
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: system 
-- Found PythonLibs: /usr/lib/arm-linux-gnueabihf/libpython3.7m.so (found version "3.7.3") 
CMake Error at /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake:141 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/boost_python-1.80.0/boost_python-config.cmake

  but it set boost_python_FOUND to FALSE so package "boost_python" is
  considered to be NOT FOUND.  Reason given by package:

  No suitable build variant has been found.

  The following variants have been tried and rejected:

  * libboost_python27.so.1.80.0 (2.7, Boost_PYTHON_VERSION=3.8)

  * libboost_python27.a (2.7, Boost_PYTHON_VERSION=3.8)

Call Stack (most recent call first):
  /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake:262 (boost_find_component)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:443 (find_package)
  kalibr/Schweizer-Messer/python_module/cmake/add_python_export_library.cmake:91 (find_package)
  kalibr/Schweizer-Messer/numpy_eigen/CMakeLists.txt:38 (add_python_export_library)

-- Configuring incomplete, errors occurred!
See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

what is next to try ?

goldbattle, 2022-09-14:
This might be related to building on an ARM platform. Are you able to try on a non-ARM platform?

zoldaten, 2022-09-16:
i tried to build on Raspbian and failed. i think that python3 version issue as i have python 3.7 but it needed 3.8.
i have successfully built it on arm Ubuntu 20.04 with ROS noetic.

zoldaten, 2022-09-29:
ok. i tried to return to Raspbian Kalibr build.
i`ve changed /usr/local/lib/cmake/boost_python-1.80.0/boost_python-config.cmake
with 
set(boost_python_FOUND 1)
and it starts building but finally crashed:
```
[ 83%] Building CXX object kalibr/aslam_incremental_calibration/incremental_calibration/CMakeFiles/incremental_calibration.dir/src/algorithms/linalg.cpp.o
make[1]: *** [CMakeFiles/Makefile2:6374: kalibr/aslam_nonparametric_estimation/bsplines_python/CMakeFiles/bsplines_python.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/algorithms/linalg.cpp:29:
/usr/include/suitesparse/spqr.hpp: In function ‘double spqr_abs(double, cholmod_common*)’:
/usr/include/suitesparse/spqr.hpp:947:51: warning: unused parameter ‘cc’ [-Wunused-parameter]
 inline double spqr_abs (double x, cholmod_common *cc)       // cc is unused
                                   ~~~~~~~~~~~~~~~~^~
/usr/include/suitesparse/spqr.hpp: In function ‘double spqr_abs(Complex, cholmod_common*)’:
/usr/include/suitesparse/spqr.hpp:952:52: warning: unused parameter ‘cc’ [-Wunused-parameter]
 inline double spqr_abs (Complex x, cholmod_common *cc)
                                    ~~~~~~~~~~~~~~~~^~
/usr/include/suitesparse/spqr.hpp: In function ‘double spqr_divide(double, double, cholmod_common*)’:
/usr/include/suitesparse/spqr.hpp:962:64: warning: unused parameter ‘cc’ [-Wunused-parameter]
 inline double spqr_divide (double a, double b, cholmod_common *cc)  // cc unused
                                                ~~~~~~~~~~~~~~~~^~
/usr/include/suitesparse/spqr.hpp: In function ‘Complex spqr_divide(Complex, Complex, cholmod_common*)’:
/usr/include/suitesparse/spqr.hpp:967:67: warning: unused parameter ‘cc’ [-Wunused-parameter]
 inline Complex spqr_divide (Complex a, Complex b, cholmod_common *cc)
                                                   ~~~~~~~~~~~~~~~~^~
/home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/algorithms/linalg.cpp: In function ‘cholmod_sparse* aslam::calibration::columnSubmatrix(cholmod_sparse*, std::ptrdiff_t, std::ptrdiff_t, cholmod_common*)’:
/home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/algorithms/linalg.cpp:66:64: error: invalid conversion from ‘std::ptrdiff_t*’ {aka ‘int*’} to ‘long int*’ [-fpermissive]
       cholmod_sparse* A_sub = cholmod_l_submatrix(A, NULL, -1, colIndices,
                                                                ^~~~~~~~~~
In file included from /usr/include/suitesparse/cholmod.h:83,
                 from /home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/algorithms/linalg.cpp:27:
/usr/include/suitesparse/SuiteSparse_config.h:62:26: note:   initializing argument 4 of ‘cholmod_sparse* cholmod_l_submatrix(cholmod_sparse*, long int*, long int, long int*, long int, int, int, cholmod_common*)’
 #define SuiteSparse_long long
/usr/include/suitesparse/cholmod_matrixops.h:192:23: note: in expansion of macro ‘SuiteSparse_long’
     SuiteSparse_long, SuiteSparse_long *, SuiteSparse_long, int, int,
                       ^~~~~~~~~~~~~~~~
/home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/algorithms/linalg.cpp: In function ‘cholmod_sparse* aslam::calibration::rowSubmatrix(cholmod_sparse*, std::ptrdiff_t, std::ptrdiff_t, cholmod_common*)’:
/home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/algorithms/linalg.cpp:98:54: error: invalid conversion from ‘std::ptrdiff_t*’ {aka ‘int*’} to ‘long int*’ [-fpermissive]
       cholmod_sparse* A_sub = cholmod_l_submatrix(A, rowIndices, numIndices,
                                                      ^~~~~~~~~~
In file included from /usr/include/suitesparse/cholmod.h:83,
                 from /home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/algorithms/linalg.cpp:27:
/usr/include/suitesparse/SuiteSparse_config.h:62:26: note:   initializing argument 2 of ‘cholmod_sparse* cholmod_l_submatrix(cholmod_sparse*, long int*, long int, long int*, long int, int, int, cholmod_common*)’
 #define SuiteSparse_long long
/usr/include/suitesparse/cholmod_matrixops.h:191:56: note: in expansion of macro ‘SuiteSparse_long’
 cholmod_sparse *cholmod_l_submatrix (cholmod_sparse *, SuiteSparse_long *,
                                                        ^~~~~~~~~~~~~~~~
/home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/algorithms/marginalize.cpp: In function ‘Eigen::MatrixXd aslam::calibration::marginalJacobian(cholmod_sparse*, cholmod_sparse*, cholmod_common*)’:
/home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/algorithms/marginalize.cpp:74:9: error: invalid conversion from ‘std::ptrdiff_t*’ {aka ‘int*’} to ‘long int*’ [-fpermissive]
         colIndices, J_x->ncol, 1, 0, cholmod);
         ^~~~~~~~~~
In file included from /usr/include/suitesparse/cholmod.h:83,
                 from /home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/include/aslam/calibration/algorithms/marginalize.h:31,
                 from /home/pi/catkin_ws/src/kalibr/aslam_incremental_calibration/incremental_calibration/src/algorithms/marginalize.cpp:19:
[... cut ...]
make[2]: *** [kalibr/aslam_incremental_calibration/incremental_calibration/CMakeFiles/incremental_calibration.dir/build.make:362: kalibr/aslam_incremental_calibration/incremental_calibration/CMakeFiles/incremental_calibration.dir/src/algorithms/marginalize.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5458: kalibr/aslam_incremental_calibration/incremental_calibration/CMakeFiles/incremental_calibration.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j3" failed
```

could you help a bit ?

goldbattle, 2022-11-24:
We don't support ARM platforms. I will leave this open if you make any progress, but it isn't something offically supported.

ElectricRCAircraftGuy, 2023-01-10:
If anyone has gotten this to build and solved this problem on an ARM platform (cross-building for an ARM target), I'd really like to know how. I just created this question on Stack Overflow: [Buildroot: `CMake Error at FindPackageHandleStandardArgs.cmake:230 (message): Could NOT find Boost (missing: thread) (found version "1.80.0")`](https://stackoverflow.com/q/75066131/4561887)

zdxzzzdx, 2024-02-25:
try this:
kalibr_workspace/src/Kalibr/Schweizer-Messer/python_module/cmake/add_python_export_library.cmake:89

list(APPEND BOOST_COMPONENTS python38)
↓
list(APPEND BOOST_COMPONENTS python3)