ethz-asl/kalibr #567: exec /bin/sh: exec format error

zoldaten, 2022-09-10:
hi!
i`m trying to build docker on raspberry pi(
ubuntu Release:        20.04
Linux version 5.4.0-1015-raspi (buildd@bos02-arm64-074) (gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2)) #15-Ubuntu SMP Fri Jul 10 05:34:24 UTC 2020)
and crash at step 2/7:
```
Step 2/7 : RUN apt-get update && DEBIAN_FRONTEND=noninteractive         apt-get install -y      git wget autoconf automake nano    python3-dev python3-pip python3-scipy python3-matplotlib        ipython3 python3-wxgtk4.0 python3-tk python3-igraph        libeigen3-dev libboost-all-dev libsuitesparse-dev  doxygen         libopencv-dev   libpoco-dev libtbb-dev libblas-dev liblapack-dev libv4l-dev        python3-catkin-tools python3-osrf-pycommon
 ---> [Warning] The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
 ---> Running in ad35f7b62fd9
exec /bin/sh: exec format error
The command '/bin/sh -c apt-get update && DEBIAN_FRONTEND=noninteractive        apt-get install -y git wget autoconf automake nano         python3-dev python3-pip python3-scipy python3-matplotlib  ipython3 python3-wxgtk4.0 python3-tk python3-igraph      libeigen3-dev libboost-all-dev libsuitesparse-dev  doxygen         libopencv-dev   libpoco-dev libtbb-dev libblas-dev liblapack-dev libv4l-dev        python3-catkin-tools python3-osrf-pycommon' returned a non-zero code: 1
```
i tried to add to docker file a first line:
`#!/bin/sh`
but i didnt help.

how to go proceed building ?

goldbattle, 2022-09-14:
It looks like you are trying to build on a `amd64` platform. I don't know if docker and do this.

zoldaten, 2022-09-16:
i have built from source without docker

closed 2022-09-16 by zoldaten

gorghino, 2023-03-01:
I had the same problem with my Jetson NX (arm64) and I changed the docker image in the Dockerfile

I used
`FROM arm64v8/ros:noetic
`