ethz-asl/kalibr #570: ROS bag containing the image and IMU data

JessicaBertolasi, 2022-09-19:
Hi, I need a suggestion on how to create correctly the bag for imu-camera calibration. How can I record the images from my video? Is it e2vid useful for this purpose? because I saw that I need to insert images and a file with imu recurds.
the data in question is `--bag filename.bag`
thank you and best regards!

goldbattle, 2022-09-19:
All details for the bag creation and extraction are on the wiki page:
https://github.com/ethz-asl/kalibr/wiki/bag-format

JessicaBertolasi, 2022-09-19:
okay but I cannot understand how to create the images. how can I extract them from an .avi video?

goldbattle, 2022-09-19:
I am not sure, I recommend googling it. Once you have them in the specified
format you can create a bag.

JessicaBertolasi, 2022-09-19:
on wiki pages there isn't an explanation on how to do it. I have a video (.raw and .avi) and also a file with `timestamp,x,y,p` from event-camera record, any suggestions?

JessicaBertolasi, 2022-09-19:
okay I've found a way to extract frames from video, but I save frames like `fname = "frame_" + str(frameId) + ".jpg"`
ideas on how to write the name with timestamp?

goldbattle, 2022-11-24:
This isn't something Kalibr supports, please try searching the internet.

closed 2022-11-24 by goldbattle