ethz-asl/kalibr #571: Multicam calibration: what is the max distance from camera to calibration target?

antoan-flox, 2022-09-22:
I'd like to determine if I can use Kalibr to calibrate extrinsics of a multi camera setup.  I am a bit unsure as to whether it would be feasible in my case as the cameras I'd like to calibrate are not situated on a robot and in close proximity.

My setup:

I have a row of 5 ceiling mounted,  fisheye cameras situated 3+ meters above ground, facing down with overlapping fields of view,  separated by about 9 meters.

Any thoughts would be greatly appreciated.

goldbattle, 2022-09-22:
If you have pairs of overlapping cameras, then there should be no issues.
You can take a look through the wiki also:
https://github.com/ethz-asl/kalibr/wiki/multiple-camera-calibration#an-example-run-using-a-sample-dataset

antoan-flox, 2022-09-23:
Thank you for reply.

I would be grateful if you could point me to somewhere that would help me determine the calibration target size that would be optimal for the distances involved in my case.

I plan to have my target moving on the ground plane facing the cameras on the ceiling which are about 3 meters high?

- Would it be necessary to make use sizes which fill the camera FOV, as I have seen mentioned in some places.
- Would the 3m vertical distance to calibration target be too large to get accurate results?

"Typically, you can achieve pose accuracy up to about 2.4 meters. Beyond this level, the results tend to vary substantially. Consequently, a robot shouldn't localize unless within the 2.4-meter threshold range."
https://www.foamcoreprint.com/apriltag-targets

Thanks again.

goldbattle, 2022-11-24:
I don't have any specific recommendation, but I think what you linked seems reasonable. It might be something that would need to be determined experimentally, as it should depend on resolution + depth to target.

closed 2022-12-06 by goldbattle