ethz-asl/kalibr #572: Wrong estimated translation IMU-Camera

ClementLeBihan, 2022-10-05:
Hi all,

I'm trying to use IMU-Camera calibration between a UEye UI3240CP and UM7 IMU. 

I did a good Intrinsic Calibration parameters with a sub-pixel reprojection RMS.
Unfortunately, I can't get good results with Kalibr for IMU-Camera Calibration. 

Here is the pdf report and a video of the rosbag I did. I hope someone will be able to help me :)

Video : https://magcloud.magellium.com/owncloud/index.php/s/7761Lf3j6OxD42L
PDF : https://magcloud.magellium.com/owncloud/index.php/s/5kGUDJS1r6hW5ek

Thank you,
Clément

goldbattle, 2022-11-24:
Hi, I was unable to download you bag.

ClementLeBihan, 2022-11-29:
Hi @goldbattle, here is a link to download the bag : https://magcloud.magellium.com/owncloud/index.php/s/Kvb5T3Dr2GkNol6

Best,
Clément

CongUcas, 2022-12-20:
> Hi @goldbattle, here is a link to download the bag : https://magcloud.magellium.com/owncloud/index.php/s/Kvb5T3Dr2GkNol6
> 
> Best, Clément

I also want to need your bag file since I met the same problem using my bag which was recorded in my office. I would like to compare them maybe I can find some valuable infomation. However the file fails to be downloaded.

ericdonatyy, 2023-01-04:
I am also facing the same problem. how was this problem solved?