ethz-asl/kalibr #579: ![微信图片_20221110220217](https://user-images.githubusercontent.com/92718554/201111878-e1602aab-2742-4df8-88c5-2bcd7cd10e23.jpg)

heisetiantang, 2022-11-10:
[image]
[image]
[image]
[image]

我与作者出现了相同的问题，我的标定板是8×6的，是大于3×3的，为什么任然出现上述问题，是否可能因为april的命名方式，必须加上a*b和每一格的大小，

I have the same problem with the author. My calibration plate is 8×6, which is larger than 3×3. Why does the above problem still occur?

_Originally posted by @heisetiantang in https://github.com/ethz-asl/kalibr/issues/370#issuecomment-1310332091_

goldbattle, 2022-11-24:
Could you provide a bag for which this fails on?

goldbattle, 2023-02-13:
Please reopen if you have the bag.

closed 2023-02-13 by goldbattle

Spring-Lee-cpu, 2023-02-24:
I meet exactly the same question.How to solve it, please

goldbattle, 2023-02-24:
Do you have a bag which reproduces this? If so can you upload it and share
it?

wbl17792615320, 2023-05-08:
I encountered the same problem, the entire Internet almost no solution, I saw someone said to rewrite "errList.append" to "print", but I still misstate SyntaxError : invalid syntax

luckydipper, 2024-07-11:
I made silly mistake, Confused sequence_time for number_of_message.
[image]
Sequence time was 53439, not 10669527

imu_info.yaml file